EdgeTX/edgetx #253: Special Function Adjust does not have additional setting line

jrwieland, 2021-06-18:
In Special Functions Trying to adjust a global variable via trim
There is no option to select what to adjust (i.e. Global Variable 1-9)

There is a switch selection
Adjust & enable

[image]

 - EdgeTX version (e.g. 2.4.0 (36870928)
 - Hardware ( RM TX16S)

raphaelcoeffic, 2021-06-18:
@jrwieland thx! You really did a hell of a testing job! Will be fixed one-by-one!

jrwieland, 2021-06-18:
The target of the adjust is still there (set prior to conversion) and works properly if you don't try and edit it.  However, if you edit it the target is lost.  and there is no way to reassign it.

commit 3baa932f (2021-06-19): Add special function parameter for "Adjust" function

Resolves #253
closed 2021-06-19 by raphaelcoeffic