EdgeTX/edgetx #255: Curve Names not saved

jrwieland, 2021-06-18:
While going though all of the settings to ensure proper conversion
The Curve names were missing from the mixes
went to curves, Cleared the name (since it would not let me edit what was converted (blank characters of some sort)
added names to the curves ( Low, Med, Hi, Bla)
checked the mixes and they were there again
Changed models and returned to the previous model
Curve names had reverted to no name ((blank characters of some sort) again could not edit without clearing

Tried 3 times/ways  to set the curve names
set curve names 
Return to main screen
Power off
edited names not saved

 - EdgeTX version ( 2.4.0 #36870928 )
 - Hardware ( RM TX16S)

jrwieland, 2021-06-18:
After clearing the curves completely and starting from scratch I did get the names and setting to save.

pfeerick, 2021-06-18:
What was the reason for restarting between each curve? I'm probably still running the RC1 build, but I was just able to add three curves in succession with unique three letter names and they were fine, and they showed up when I went both the input and mixer settings... this was on a new model though, not an imported one though, so it's either conversion related still or a regression I haven't yet...

raphaelcoeffic, 2021-06-18:
> What was the reason for restarting between each curve? I'm probably still running the RC1 build, but I was just able to add three curves in succession with unique three letter names and they were [...]

My gut feeling tells me this is a conversion issue.

jrwieland, 2021-06-18:
@raphaelcoeffic your gut is correct it is a conversion issue.  I get to load a fresh version of Edge tonight (related to issue #253) so I'll take screenshots to show what is happening with the names and actual curves on conversion.  
Converted models are from V. 2.3.12.

jrwieland, 2021-06-19:
ok here is what is happening to the curves 
1.  install edgeTX,/SD Card Conversion takes place
2. Go to curves (no Names, values are correct though)
[image]
3.  Name curves 1-4
[image]
 4.  Check the mixes - everything looks good
 5. Power off the radio and restart
 6. Names are gone and mix values have been changed on all curves edited prior to the power cycle
 [image]

commit afe7c13b (2021-06-19): Fix conversion of input line name, curve names, and write radio.bin to disk early

Resolves #264
Resolves #255
closed 2021-06-19 by raphaelcoeffic