EdgeTX/edgetx #256: Widget background color has no effect

eshifri, 2021-06-18:
**Describe the bug**
Changing Widget background color has no effect.

**General information**
 - EdgeTX version 2.4.0 current code compiled.
 - Hardware RM TX16S

pfeerick, 2021-06-18:
What are the steps? How do you change the widget background colour?

eshifri, 2021-06-18:
Push "TELEMETRY" button -> push "Next Page" twice (we are on the second telemetry screen) -> select 2x6 widget template.
There will be options to change background color of two columns independently.  
This option does not have any effect.

pfeerick, 2021-06-18:
Oh, _noice_... layout background colours! And yeah, I'm see that also... There are options for Panel 1 and Panel 2 backgrounds, for the 2x4 layout, but they don't seem to do anything. 

[image]
[image]

eshifri, 2021-06-19:
Looks like widget::background in widget.h does not have implementation:
virtual void background() {}

raphaelcoeffic, 2021-06-19:
> Looks like widget::background in widget.h does not have implementation:
> virtual void background() {}

Sorry, this one is for something else ;-) You should have a look at the layout itself. You should be able to override `paint()` and thus get what you need ;-)

eshifri, 2021-06-19:
Yes,  thank you, I have realized this, but I still cannot figure whow the Layout is refreshed/painted. In the NV14 version there was a refresh() function - here I do not see anything for drawing in the Layout.

raphaelcoeffic, 2021-06-19:
All the layouts are inherited from this:
```
class WidgetsContainer: public FormGroup
```

So basically, as any `Window` (`FormGroup` is a window), you can just override `void paint(BitmapBuffer* dc)`.

eshifri, 2021-06-19:
Thank you! This was helpful.
A pull request for the fix created. :-)

pfeerick, 2021-07-29:
Closed by #272

closed 2021-07-29 by pfeerick