EdgeTX/edgetx #260: R9 ACCESS MOD, R9 FLEX and Bluetooth Mod support on Horus Type radios

b14ckyy, 2021-06-18:
As we are not able yet to use the OpenTX Companion, we have no option to enable and disable firmware features. 
So as the title says, the following features are missing: 

FrSky R9 FLEX Firmware support (all radios I guess)
FrSky R9 ACCESS MOD support (For Horus Type radios that have the Mod installed to use UART1 internally for ACCESS Protocol support)
Bluetooth Mod support (For Horus Type radios that have the Horus X10S Bluetooth Module installed, using UART2 for internal communication like trainer and telemetry outoput)

I really love EdgeTX afte a few minutes of playing around with it but without these it is not yet usable, at least for me.

pfeerick, 2021-06-18:
I'm not familiar with those mods, but from how you describe it it sounds like the options you would tick off in the companion under 'build options'. This is possible to do without the companion, but is obviously not user friendly, as it requires setting up a build toolchain, or installing docker and using a docker container to build the firmware with a given set of flags. 

So I can fully understand you position, and this is something we intend to work on post 2.4.0 being released, as the current situation is indeed not idea.

In the mean time, if there was a specific firmware flavor you wanted, I'm sure anyone with a working build setup would be more than happy to run off a firmware build with the specific options you need - you'd just need to specify the radio type, menu language and build options.

raphaelcoeffic, 2021-06-18:
Build options require an online build platform which we will be building once we have a stable release.

b14ckyy, 2021-06-18:
Oh Awesome! Then I am ready :D

raphaelcoeffic, 2021-06-18:
> Oh Awesome! Then I am ready :D

But if I were you, I'd prefer using ELRS on the same hardware: works much better, further, and with lower latency.

b14ckyy, 2021-06-18:
> > Oh Awesome! Then I am ready :D
> 
> But if I were you, I'd prefer using ELRS on the same hardware: works much better, further, and with lower latency.

Nahhhh don't drag me into that kind of discussion. ELRS is absolutely no option for me ;) IMHO ELRS should not even exist. But that's another topic and not for this thread ;)

JyeSmith, 2021-06-18:
@b14ckyy you got my interest.  Why shouldn't ELRS exist?

DaySpeed955i, 2021-06-18:
I have a X12S ISRM with externalACCESS mod.

Once installed EdgeTX is working well, conversion was good except for THR channel, it took ch10 instead of input THR.

One more problem that might be related to no ACCESS support for now is that when I turn off my X12S, it always goes into emergency mode, then the STM32 bootloader is not recognized by Windows, I have to remove the Tx battery for more then 5 minutes to have the bootloader from the Tx to be functional again.
If I turn ON the X12S, it starts normally (I had that too with an early version of ETHOS).

I can then load OTX or ETHOS or EdgeTX again.

Thanks for all you work into EdgeTX.

brainbubblersbest, 2021-06-20:
> @b14ckyy you got my interest. Why shouldn't ELRS exist?

Because In Europe you get in lot of trouble by using not the definitions of the ISM Band.
Mobile Phone Carriers can sue you for Transmitting without taking Care of Duty Cycle rules and maybe for Blocking A Phone Call for Emergency Reasons. (Blocking Calls to Medic or Police etc and mybe someone died for that is not so Funny for the Law Enforcement)

pfeerick, 2021-06-21:
Isn't that an issue of using the wrong frequencies rather than a problem with ELRS? It's not the problem of the ELRS project if you don't use the correct ISM frequencies in your part of the world. i.e. Here in Australia one of the ISM bands is 915Mhz, which does not overlap with any of the mobile phone frequencies here, by design.

brainbubblersbest, 2021-06-21:
https://www.disk91.com/2017/technology/sigfox/all-what-you-need-to-know-about-regulation-on-rf-868mhz-for-lpwan/

And why 500hz refreshrate is hard to believe with 868mhz. Even with very short bursts 915 has no duty cycle. But your in the carrier Band.

raphaelcoeffic, 2021-06-21:
> https://www.disk91.com/2017/technology/sigfox/all-what-you-need-to-know-about-regulation-on-rf-868mhz-for-lpwan/
> 
> And why 500hz refreshrate is hard to believe with 868mhz. Even with very short bursts.
> 915 has no duty cycle. But your in the phone carrier Band.
> 
> Edit: my Post before was maybe a little unclear.
> Fact is.
> Legal r9 on 868mhz with 500mw has no telemetry because of that.
> (unidirectional on 500mw)
> But without duty cycle
> 
> And with telemetry you have only 1/20 Part from that.
> And you have the Duty Cycle Rule.
> Thats legal use in Europe. 500hz wit 36sec/h transmit Time is hard to believe.

The point here is: the 868Mhz band in Europe is indeed very much constrained, that's true. Not so much at other places. So this is a european issue, more than anything else.

b14ckyy, 2021-06-21:
Fact is: ELRS has no certification and is illegal basically everywhere around the globe. Not just in Europe. 
Refresh rates that high are useless on long range flights and for short range within a few km 2.4GHz is perfectly fine. 
It does never LBT so it causes micro failsafes for other RC Systems in the band.
And all that fiddling around with telemetry rates to find a sweet spot where it works with telemetry applications without losing too mjuch range is also a hazzle. 

And compared to modern R9 Hardware or Crossfire it has ABSOLUTELY ZERO benefits. All 3 systems will outperform any video link, no matter what. And I don't care if I have to be on 1000mW to get to 50 or even 100km or if it can do it on 10mW. As long it is reliable and gives me telemetry as long as possible. 

@JyeSmith you wanted my opinion, here is it.

brainbubblersbest, 2021-06-21:
What can i say, You both are right 😅

pfeerick, 2021-06-23:
> 915 has no duty cycle. But your in the phone carrier Band.

Because we're in different ITU regions, we'll both be talking about things that are wrong for our regions. As here 915Mhz *does* have a duty cycle, and it's also not in the phone carrier band. Using the different ISM spectrum is done deliberately, so that certification isn't needed. If you use an exempt class license band, you don't need certification. 

I think the real problem is the duty cycle and LBT European "problem" (not so for the rest of the world, yet?). You only have to look the TTN's spectrum plan for LORA, to see the different spectrum plans, duty cycle and other requirements in the various regions around the world, and it is an absolute mess. It's the main reason 2.4Ghz has been used, because of the worldwide consistency, not because it's the best or the right frequency. I also don't get what's with the ridiculously high refresh rates... we just can't move the sticks or servos fast enough to be pushing for those rates for LOS flights, let alone long range stuff. 

Anyway, we're (I'm?) derailing this issue, so as much as I'd like to be able to discuss it more, lets have it resume to the title topic... ;)

CorruptName, 2021-06-29:
I'm going to close my issue since I was literally asking for the same 3 things.

CorruptName, 2021-06-29:
I too would like to be able to use custom build flags.
I'm on a TX16S and use the following build flags:
`flexr9m
bluetooth
externalaccessmod`

Has anyone compiled for TX16S with these flags so I don't have to set up a toolchain?

b14ckyy, 2021-06-29:
@pfeerick has compiled one for me and also someone else with the correct flags. But R9 ACCESS does not seem to be implemented correctly. Binding is not working (register works). The binding popup disappears immediately.

CorruptName, 2021-06-29:
> @pfeerick has compiled one for me and also someone else with the correct flags. But R9 ACCESS does not seem to be implemented correctly. Binding is not working (register works). The binding popup disappears immediately.

Oh, that sucks. Care to post it so I can test it out?

b14ckyy, 2021-06-29:
sure: 
[EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip](https://github.com/EdgeTX/edgetx/files/6736006/EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip)

CorruptName, 2021-06-29:
> sure:
> [EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip](https://github.com/EdgeTX/edgetx/files/6736006/EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip)

Thanks. I'll play around with it and report back.

AlessandroAU, 2021-06-30:
With regards to duty cycle limitations and for some perspective, this is what crossfire looks like on 868 CE, make up your own minds if that complies with the 1% or 5% or whatever are the duty limits in EU. Also, lets not forget that r9 flex is uncertified, so are those SPI whoop RX implementations, MPM and a whole lot of other things... (Where's the FCC sticker for my GHST TX btw?)  

[image]

pfeerick, 2021-06-30:
Don't you also have to set the region to CE or FCC as needed, rather than Open? As the frequency used is only half of the equation. I would still expect the duty cycle to be out though...

b14ckyy, 2021-06-30:
> With regards to duty cycle limitations and for some perspective, this is what crossfire looks like on 868 CE, make up your own minds if that complies with the 1% or 5% or whatever are the duty [...]

can you check the same with ELRS 868/915 please? 

Also when I look at your waterfall, I see Crossfire hopping between around 50 different channels there. That means a duty cycle of 2%. was a receiver turned on with Telemetry enabled? And you are extremely close to the source so the Noise on the band is expected.

AlessandroAU, 2021-06-30:
Hmm I very much doubt 50 channels. Looks like 500khz BW, so about 20 channels since the band is ~10 MHz, This is consistent with sx1276 modulation settings. Also, what noise?, my noise floor is a steady -80 and the packets are about -30.

Original pic was without RX turned on, here is is enabled for both systems. 

[image]

b14ckyy, 2021-06-30:
I have counted the channels by your picture but it is slightly distorted and I just counted to the center. This is 25 channels marked at the center points. Of course they are overlapping. So the total number of channels are between 40-50. 
[image]

And now look at your comparison between crossfire and ELRS on similar update rates. Crossfire has much more channel hoppings and also looks like shorter time per channel. If course I can't calculate the precise duty cycle by these waterfalls. 

The Bandwith looks similar though. I know in Race mode Crossfire widens the bandwith to overcome interferences but on 868CE it should be fine. At least here ELRS seems to use the same channel width. Mit not be that problematic then.

But your analysis is really nice! Thanks for your work. Would be great if you could do more measurements in different Signal rates on both. I am happy to get proven wrong with my first estimations. I will get a ELRS 2.4GHz setup soon and will also do some checks.

pfeerick, 2021-07-01:
I'd also like to know if the region settings have any impact if you get a chance to look at them... since the crossfire manual indicates: 
[image]

I don't know if it *only* locks in the frequency and power, or also enables things like LBT and other limiters (i.e. duty cycle) in order to be compliant, which it would not do if you are have it set to open. Unless they are faking it, I would have expected Crossfire to be EU compliant as far as duty cycle, etc when in CE mode since it's RED certified.

AlessandroAU, 2021-07-01:
I set the region to CE on those tests, it changed nothing. As the manual says it just locks you out of the other options, the options themselves don't have any different behaviour. It is not surprising to that CRSF does not obey the strict duty cycle limits of EU, a control link would not be feasible with those limitations. There's a post somewhere on Facebook when trappy says you can 'use up to 50% duty' under some interpretation of the rules. I couldn't find it right now.

BTW, that RED certification is self-signed by TBS and NOT verified by a third party as far as I can see. Don't forget they were selling some hardware for years without FCC approval, look at the FCC listing for the nanoRX for example, it's dated 2020.

pfeerick, 2021-07-01:
Thank you for testing, I just wanted to make sure, because excerpts like I showed from the manual don't always tell you *everything* that happens. 

Yes, it is self-signed, but I'm sure that opens them up to hefty fines if the regulators find out it was a false claim, so verification or not, its not something you want to do if you want to stay in business. If they have thought it through or not, I can't say, it just doesn't make any rational long-term sense. 

I fully agree... I find it hard to believe that with a strict 5% duty cycle it would be possible to hold a control link... and some very interesting arguments can be made when you read through the various rules... I've also seen some that suggest that by walking the spectrum it's 5% in relation to the given frequencies it transmits on, so you only need... what... 20 frequency steps because it's spread spectrum, not fixed frequencies, and you have 100% duty cycle?

MartyFufkin, 2021-07-12:
Interesting thread...  I wish I had read it before getting too keen on EdgeTx.  I have other radios I can test with but I'll be putting my daily use TX16S back to OpenTX until R9M Access is available on this platform.  Too invested in it to care about any CRSF pro/con... 

I'll be checking back often.  Great work to all the developers and contributors.   EdgeTX is looking good so far.  
Over and Out.

Marty

brainbubblersbest, 2021-07-12:
It's been working on. Thats been All Release Candidates so far.

MartyFufkin, 2021-07-12:
ok...  I'll give it another look.  I could not find options for Flashing OTA for R9 Access and didn't manage to get a bind working when migrating from OpenTx (bind data seemed to reset).  I will have another look.   When using External R9M 2019 with Access firmware and internal mods, there are no options for Registering the R9M Receiver...  Am I missing something?

MartyFufkin, 2021-07-12:
I think the problem is my end...  Revision control issue.  I think I may have an earlier Nightly Build addressing a previous post I made.  I will move the 2.4 RC3 and check for Access options and report back.

brainbubblersbest, 2021-07-12:
Need to compile with 
-DHARDWARE_EXTERNAL_ACCESS_MOD=YES
As long there is no True Release with working companion. 
And the Module options are not selectable.

MartyFufkin, 2021-07-12:
@brainbubblersbest ...  Thanks.  I compiled as per your suggestion and R9 Access now appears as an option for External Module.  Haven't tested for functionality yet but will soon,.  Thanks again for your help.

MRC3742, 2021-09-20:
The external access mod is not entirely working on the Jumper T16

When selecting External RF "R9M ACCESS" the two options are "FCC" or "EU". Should there be a selection for "FLEX" ?

Current build of EdgeTX is able to register the RX successfully but then unable to bind using any of the three Bind selections.
The Bind screen opens momentarily then closes immediately without any confirmation of RX.

Also there does not seem to be any "Options" setting to change the Module Power Levels.

OpenTX 2.3.14 Screen
[image]
EdgeTX 09/20/21 build
[image]

I am currently running newest Flex firmware on the TX Module and RX.

Thanks - Rich

raphaelcoeffic, 2021-09-20:
> The external access mod is not entirely working on the Jumper T16
> 
> When selecting External RF "R9M ACCESS" the two options are "FCC" or "EU". Should there be a selection for "FLEX" ?
> 
> (1) Current build of EdgeTX is able to register the RX successfully but then unable to bind using any of the three Bind selections.
> The Bind screen opens momentarily then closes immediately without any confirmation of RX.
> 
> (2) There is no "Receiver No." option available
> 
> (3) There is no "Options" setting to change the Module Power Levels.
> 
> OpenTX 2.3.14 Screen
> [image]
> EdgeTX 09/19/21 build
> [image]
> 
> I am currently running newest Flex firmware on the TX Module and RX.
> 
> Thanks - Rich

Did you try to add the corresponding options for FLEX?

MRC3742, 2021-09-20:
Yes, I beleive I did.
I compiled with options
-DHARDWARE_EXTERNAL_ACCESS_MOD=YES
-DMODULE_PROTOCOL_FLEX=YES

Rich

MRC3742, 2021-09-20:
This is the Version Screen showing the Options

[image]

brainbubblersbest, 2021-09-21:
I can copy that. Same compile options.
No flex available.
And no module options available.
Same with internalaccess. No module options.
Flashed flex. But it appears only selectable with r9m accst.

raphaelcoeffic, 2021-09-21:
> I can copy that. Same compile options.
> No flex available.
> And no module options available.
> [options] [set]
> Same with internalaccess. No module options.
> Flashed flex. But it appears only selectable with r9m accst.
> X12s.

Gentlemen, I’ll need some help from your side, as I absolutely no R9 access hardware to test.

janmj, 2021-11-15:
> Gentlemen, I’ll need some help from your side, as I absolutely no R9 access hardware to test.

I just noticed this issue the other day when i was testing EdgeTX on my TX16S. the "915" and "816" choices, indicating FLEX mode is missing in the menu on the radio. However the choices are there in companion.
What can i do to help debug this?

pfeerick, 2021-11-15:
Most likely try out https://github.com/EdgeTX/edgetx/pull/1043 - you should be able to get the firmware via the Checks tab -> 
Run tests and package firmware and then look under the Artifacts heading for the zip file with the firmware package.

janmj, 2021-11-15:
> Most likely try out #1043 - you should be able to get the firmware via the Checks tab -> Run tests and package firmware and then look under the Artifacts heading for the zip file with the firmware package.

That seems to work just fine. The "915" and "868" choises are back, and is also carried over from OTX converted models.
I will try to bind with e receiver later today.

b14ckyy, 2021-11-15:
> Most likely try out #1043 - you should be able to get the firmware via the Checks tab -> Run tests and package firmware and then look under the Artifacts heading for the zip file with the firmware package.

Dies this also include ACCESSMOD support on non FrSky Radios or modded FrSky radios?

janmj, 2021-11-16:
So now i have tested binding a receiver (R9MM), it boud fine, but the radio would not stop chirping after i exit out of bind mode. Also when i have chosen R9M - 868/915 for flex mode, i can no longer choose the transmit power.

ucakar, 2021-11-19:
> sure:
> [EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip](https://github.com/EdgeTX/edgetx/files/6736006/EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip)

Hi is it possible to get EdgeTX 2.5 for Jumper T18 with: LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_R9M support

DM4DS, 2021-11-19:
> > sure:
> > [EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip](https://github.com/EdgeTX/edgetx/files/6736006/EdgeTX_TX16S_LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_5d9498e.zip)
> 
> Hi is it possible to get firmware of EdgeTX 2.5 for Jumper T18 with: LUA_BLUETOOTH_EXTERNAL_ACCESS_MOD_R9M support

https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Windows-10-%28MSYS2%29#setting-up-the-build-environment-for-edgetx

Had to do it too while I have bluetooth mod in my t18....
If you can´t get it run and ok with ELRS 2,.6 I can send you my file, I´m using.

ucakar, 2021-11-19:
Yes, please send me your file. thx!

ucakar, 2021-11-19:
I have Jumper T18 Pro, bluetooth and External Frsky R9 2019 module.

DM4DS, 2021-11-20:
> Yes, please send me your file. thx!

While I think you are not the last , I uploaded the file to my homepage and will upload always the last version on my dialy, NOT the experimental versions I´m testing on my second. so you can come back and look there.

https://www.dm4ds.de/

Hope this help.

ucakar, 2021-11-20:
Hi DM4DS,

I am looking for externallaccessmod and flexr9m. Can you please upload that option too?

Thanks,

Andrej

DM4DS, 2021-11-20:
> Hi DM4DS,
> 
> I am looking for externallaccessmod and flexr9m. Can you please upload that option too?
> 
> Thanks,
> 
> Andrej

Uploaded an untested version to my homepage with EXTERNAL and Flex enabled - give me feedback on my discord

Anduurfly, 2021-12-07:
> > Hi DM4DS,
> > I am looking for externallaccessmod and flexr9m. Can you please upload that option too?
> > Thanks,
> > Andrej
> 
> Uploaded an untested version to my homepage with EXTERNAL and Flex enabled - give me feedback on my discord

Hi,
I am now testing the latest version you created for TX16s (wiring mode was already done). Thanks a lot DM4DS. First i flashed the radio, then i flashed the r9m 2019 access module with the v1.3.0 ACCESS FLEX firmware. 
Flashed an r9 mini with v1.3.1 ACCESS FLEX firmware. 
The registration and binding was ok using the 868 protocol. The receiver's led was green.
But there were no output on S.Bus
I tried with an r9 SX also on ACCESS FLEX firmware.  The receiver's led was green.
But there were no output on any channel.
I re-flashed the whole staff to LBT firmware (i am in EU). Set the external TX to R9ACCESS and EU.
The result was the same.
Then by some intuity i set the radio to R9ACCESS FCC without changing anything else. The receivers started working immediately.
I made some more experiments and found whatever firmware pairs i have on the r9m and the receivers they are binding well, but i get channels outputs only if i set the radio to FCC even all has LBT firmwares.
Is that a known issue? Anyone else experienced the same?
I suppose if i use LBT formware the communication would be 868mhz whatever i set right?
Thanks
Andras

DM4DS, 2021-12-09:
HiI don't use any R9 hardware. The problems you are writing are not known to me.Have you asked in the EdgeTX discord? Maybe it's a big in the firmware, setting a flag incorrect. Just an idea.Sorry, can't help you.GreetingsSandor-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

brainbubblersbest, 2021-12-09:
Firmware on tx module need to be flashed to eulbt to use it in eulbt mode

Anduurfly, 2021-12-09:
Hello Sandor, thank you for your feedback and your work on this. I keep investigating.

@brainbubblersbest , did you have experience wit this edgetx version and the r9? If you read my description i wrote that i tried with flex and LBT firmwares too. (flex-flex on the module and the receiver // lbt-lbt on the module and the receiver)
But in both cases i get output signal from the receiver if i set the external tx to FCC. This was my problem. 
I believe if i use the LBT firmware it wil define the frequency itself whatever is set on edgetx.
If you have the same hw setup your experience is more than wecome.
Thanks

brainbubblersbest, 2021-12-10:
I will try eulbt.

jpmarotta, 2021-12-31:
I can confirm that this issue also occurs on a mod'd TX16S which previously worked flawlessly on OTX.
Same symptoms as above: able to "register", unable to bind.

Tested with 2019 R9M on FCC and FLEX firmware.

The following RXs were included in the testing in both FCC and FLEX:
- R9MX OTA
- R9 Slim+ OTA
- R9 STAB

Here's my build if it's useful:
[edgetx-v2.4_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip](https://github.com/EdgeTX/edgetx/files/7795276/edgetx-v2.4_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip)

b14ckyy, 2021-12-31:
> I can confirm that this issue also occurs on a mod'd TX16S which previously worked flawlessly on OTX. Same symptoms as above: able to "register", unable to bind.
> 
> Tested with 2019 R9M on FCC and FLEX firmware.
> 
> The following RXs were included in the testing in both FCC and FLEX:
> 
> * R9MX OTA
> * R9 Slim+ OTA
> * R9 STAB
> 
> Here's my build if it's useful: [edgetx-v2.5_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip](https://github.com/EdgeTX/edgetx/files/7796661/edgetx-v2.5_tx16s_lua-ppmus-mode2-R9M-ACCESS-noheli_release.bin.zip)

all is fixed in the nighlies. I tried it recently and ACCESS works flawlessly including FLEX

jpmarotta, 2021-12-31:
> all is fixed in the nighlies. I tried it recently and ACCESS works flawlessly including FLEX

Just built and tested, works as expected. Looks like I'm staying on ETX. Thanks for the heads up!

[35 more replies]

closed 2022-12-10 by b14ckyy