EdgeTX/edgetx #264: Conversion from OTX 2.3.12 to EDGETX 2.4.0 loses input line names

lshems, 2021-06-18:
When converting from OTX 2.3.12 to EDGETX 2.4.0, the line names are lost in conversion.

Every input has an 'inputName' attribute.
For every line belonging to an input, there is the attribute 'name', that becomes cleared o, conversion.

raphaelcoeffic, 2021-06-19:
> When converting from OTX 2.3.12 to EDGETX 2.4.0, the line names are lost in conversion.
> 
> Every input has an 'inputName' attribute.
> For every line belonging to an input, there is the attribute 'name', that becomes cleared o, conversion.

Could you please please add an example model file (incl. models.txt, and radio.bin), that would help greatly to debug it.

lshems, 2021-06-19:
can we add OTX as permitted file type, by the way?

[t16_23.zip](https://github.com/EdgeTX/edgetx/files/6680426/t16_23.zip)

raphaelcoeffic, 2021-06-19:
Thx for the files! The OTX support can be discussed in your new issue.

raphaelcoeffic, 2021-06-19:
> For every line belonging to an input, there is the attribute 'name', that becomes cleared o, conversion.

I do see some of them, could you please tell me which ones are missing?

[image]

commit afe7c13b (2021-06-19): Fix conversion of input line name, curve names, and write radio.bin to disk early

Resolves #264
Resolves #255
closed 2021-06-19 by raphaelcoeffic