EdgeTX/edgetx #266: Outputs widget shows wrong channel order.

KevinFairgrieve, 2021-06-19:
Vers: 2.4.0 (952bc12b)
Date: 2021-06-18
Time: 08:26:36
EEPR: 220

I have my hardware set up to TAER.

When I add a widget to the screen as I do for nearly all my models. The Outputs widget shows the stick order as RETA.

Under OTX this showed what the hardware is set to.

Is this a bug or intentional? I for one would rather the widget showed what the hardware was set to.
[image]
[image]
[image]

lshems, 2021-06-19:
The RETA option only defines the default INPUT and OUTPUT order when setting up a new model.

Change the order of your mixer lines to what you want.

If you created a NEW model with RETA activated, and this is shown, it is a bug on the CREATION wizard, not on the outputs widget.

lshems, 2021-06-19:
Stick order is set with MODE in a radio level.
Mode 2 is left stick rudder throttle, right stick elevator aileron.

Is indépendant of RETA.

lshems, 2021-06-19:
see also
Can we get rid of the last functional legacy terminology of MODE and RETA definitions? #67

raphaelcoeffic, 2021-06-19:
> see also
> Can we get rid of the last functional legacy terminology of MODE and RETA definitions? #67

What is 100% sure is that we need to ask these settings on first boot / factory reset, no matter what the names are. We can even ask which names you prefer and set custom names for you.

closed 2021-06-19 by raphaelcoeffic