EdgeTX/edgetx #267: allow .otx as they are .zip anyway, but it would make life easier

lshems, 2021-06-19:
.otx fies are zip files, but not allowed to upload. Could we change that?

raphaelcoeffic, 2021-06-19:
I had made some experiments a couple of years ago with this, using miniz as the ZIP library. Unfortunately, it seems miniz uses the stack extensively, which is quite an issue for us. But as I remember, that was only when using compression. It should possible to use ZIP without compression to solve the issue. I’ll try to dig out my experiments when I get to it.

but what is clear already now: this won’t be used to store the models all the time, but only when importing / exporting, as it is not fast enough to allow it to be used all the time.

lshems, 2021-06-19:
LOL. I was just talking about the github issues here. I want to upoload an OTX now and then to an issue, and OTX is not allowed, only zip. Same for .lua perhaps? Culd we allow github to upload those??

raphaelcoeffic, 2021-06-19:
> LOL. I was just talking about the github issues here. I want to upoload an OTX now and then to an issue, and OTX is not allowed, only zip. Same for .lua perhaps? Culd we allow github to upload those??

Ah ok! Well maybe, but I would not know how.

pfeerick, 2021-06-21:
I don't believe we have any control whatsoever over the file types allowed in github issues. The list of supported types and sizes is [here](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/attaching-files). 

Probably the the best we can do is to recommend (seemingly pointlessly) is to zip up the otx file, so it's now a compatible file extension - most people will be able to manage that - less will know about how to changing the file extension - i.e. turning extensions on to add the extension if their OS hides file extensions, what is an extension, etc.

lshems, 2021-06-21:
Ok, if it cannot be done, then we can close the issue

closed 2021-06-21 by lshems