EdgeTX/edgetx #270: Tbs Crossfire reconnect

Capcom58, 2021-06-19:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 
Hello i noticed every time i swapp a lipo pack the crossfire module reconnects briefly then disconnects i have to completely remove the module and put it back to get it back working

raphaelcoeffic, 2021-06-19:
@Capcom58 can you please describe a bit more? Maybe edit to tell us which hardware, which firmware version for the crossfire module, etc. I'm a bit puzzled because we have quite some folks using crossfire here.

Capcom58, 2021-06-19:
yeah i was thinking the same it could just be a isolated issue im running the latest nightly crossfire micro latest firmware also Tx16s  i may completely reflash and see what happens

JimB40, 2021-06-20:
What exact CRSF Module version you use and what FW version is flashed into it?

Capcom58, 2021-06-20:
I Think its all good now after the reflash everything works

closed 2021-06-20 by raphaelcoeffic