EdgeTX/edgetx #274: Some more instructions on building on ubuntu would be helpful

lshems, 2021-06-19:
I used the nice guide in the WIKI. It all works to build my own stuff!!

Thanks.

But I'm a little bit stuck on finding a workflow to download a new version from github, without hving to recreate all directories and settings. 

I tried this, but that won't work as the directory is not empty. How do you guys have your workflow set up for testing new versions?

lshems@Khalil-Surface:~/edgetx$ git clone --recursive -b 2.4 https://github.com/EdgeTX/edgetx.git edgetx-v2.4
fatal: destination path 'edgetx-v2.4' already exists and is not an empty directory.
lshems@Khalil-Surface:~/edgetx$

pfeerick, 2021-06-21:
New version of what?

To update an existing repo, you would want to do a `git pull`, not a new clone. Better yet, do a `git fetch` to check for updates, without applying them, then you can do a `git status` which will tell you if it can be updated safely (i.e. it will say it can it is out of date, and can be fast-forwarded, and then `git pull` to actually pull/apply the updates.

lshems, 2021-06-21:
Thanks, this is very helpful.

I'm busy on a simple work flow guide for noobs. I got enough things working to get an initial guide running.
Then we can perhaps tweak it where things pop up.

I would really hope we can get a community effort going for edge which is not limited to people capable of coding. There is so much more to do.
But without a noob's guide on how to do a basic thing as creating a firmware for a specific pull request to test, this will never happen.

I will close the issue, as I got just enough of a hint to get further :).

closed 2021-06-21 by lshems