EdgeTX/edgetx #275: TEXT_INVERTED_BGCOLOR missing as constant

lshems, 2021-06-19:
T16 latest version as of today

The TEXT_INVERTED_BGCOLOR constant is nil.

jfrickmann, 2021-07-31:
I do not see such a color defined in the system. Only the following in /radio/src/gui/colorlcd/colors.h:
```
#define ALARM_COLOR                    COLOR(ALARM_COLOR_INDEX)
#define BARGRAPH1_COLOR                COLOR(BARGRAPH1_COLOR_INDEX)
#define BARGRAPH2_COLOR                COLOR(BARGRAPH2_COLOR_INDEX)
#define BARGRAPH_BGCOLOR               COLOR(BARGRAPH_BGCOLOR_INDEX)
#define BATTERY_CHARGE_COLOR           COLOR(BATTERY_CHARGE_COLOR_INDEX)
#define CHECKBOX_COLOR                 COLOR(CHECKBOX_COLOR_INDEX)
#define CURVE_AXIS_COLOR               COLOR(CURVE_AXIS_COLOR_INDEX)
#define CURVE_COLOR                    COLOR(CURVE_COLOR_INDEX)
#define CURVE_CURSOR_COLOR             COLOR(CURVE_CURSOR_COLOR_INDEX)
#define CUSTOM_COLOR                   COLOR(CUSTOM_COLOR_INDEX)
#define DEFAULT_BGCOLOR                COLOR(DEFAULT_BGCOLOR_INDEX)
#define DEFAULT_COLOR                  COLOR(DEFAULT_COLOR_INDEX)
#define DISABLE_COLOR                  COLOR(DISABLE_COLOR_INDEX)
#define EDIT_MARKER_COLOR              COLOR(EDIT_MARKER_COLOR_INDEX)
#define FIELD_BGCOLOR                  COLOR(FIELD_BGCOLOR_INDEX)
#define FIELD_FRAME_COLOR              COLOR(FIELD_FRAME_COLOR_INDEX)
#define FOCUS_BGCOLOR                  COLOR(FOCUS_BGCOLOR_INDEX)
#define FOCUS_COLOR                    COLOR(FOCUS_COLOR_INDEX)
#define HEADER_BGCOLOR                 COLOR(HEADER_BGCOLOR_INDEX)
#define HEADER_COLOR                   COLOR(HEADER_COLOR_INDEX)
#define HEADER_CURRENT_BGCOLOR         COLOR(HEADER_CURRENT_BGCOLOR_INDEX)
#define HEADER_ICON_BGCOLOR            COLOR(HEADER_ICON_BGCOLOR_INDEX)
#define HIGHLIGHT_COLOR                COLOR(HIGHLIGHT_COLOR_INDEX)
#define LINE_COLOR                     COLOR(LINE_COLOR_INDEX)
#define MAINVIEW_GRAPHICS_COLOR        COLOR(MAINVIEW_GRAPHICS_COLOR_INDEX)
#define MAINVIEW_PANES_COLOR           COLOR(MAINVIEW_PANES_COLOR_INDEX)
#define MENU_BGCOLOR                   COLOR(MENU_BGCOLOR_INDEX)
#define MENU_COLOR                     COLOR(MENU_COLOR_INDEX)
#define MENU_HIGHLIGHT_BGCOLOR         COLOR(MENU_HIGHLIGHT_BGCOLOR_INDEX)
#define MENU_HIGHLIGHT_COLOR           COLOR(MENU_HIGHLIGHT_COLOR_INDEX)
#define MENU_LINE_COLOR                COLOR(MENU_LINE_COLOR_INDEX)
#define MENU_TITLE_DISABLE_COLOR       COLOR(MENU_TITLE_DISABLE_COLOR_INDEX)
#define OVERLAY_COLOR                  COLOR(OVERLAY_COLOR_INDEX)
#define SCROLLBAR_COLOR                COLOR(SCROLLBAR_COLOR_INDEX)
#define TABLE_BGCOLOR                  COLOR(TABLE_BGCOLOR_INDEX)
#define TABLE_HEADER_BGCOLOR           COLOR(TABLE_HEADER_BGCOLOR_INDEX)
#define TEXT_DISABLE_COLOR             COLOR(TEXT_DISABLE_COLOR_INDEX)
#define TEXT_STATUSBAR_COLOR           COLOR(TEXT_STATUSBAR_COLOR_INDEX)
#define TITLE_BGCOLOR                  COLOR(TITLE_BGCOLOR_INDEX)
#define TRIM_BGCOLOR                   COLOR(TRIM_BGCOLOR_INDEX)
#define TRIM_SHADOW_COLOR              COLOR(TRIM_SHADOW_COLOR_INDEX)
```

pfeerick, 2021-07-31:
Hm, it is listed as supposed to be there per the lua doc though:
https://github.com/EdgeTX/edgetx/blob/6e37d11710f9c1a5db028dbc52af501b9790448f/radio/src/lua/api_colorlcd.cpp#L768

If you compare the list with OTX, there's quite a bit of difference...

jfrickmann, 2021-08-01:
Now the question is: do we want to change the documentation to fit with what is currently available, or do we want to add any color back that was ever mentioned in a version of the documentation? 
As you can hear, I favour the latter option and would prefer that we move ahead and support Robert's revision of theme colors in #506.

raphaelcoeffic, 2021-08-01:
> As you can hear, I favour the latter option and would prefer that we move ahead and support Robert's revision of theme colors in #506.

+1!

pfeerick, 2021-08-01:
> Now the question is: do we want to change the documentation to fit with what is currently available, or do we want to add any color back that was ever mentioned in a version of the documentation? [...]

Any version? No - at best, 2.3-2.4. And since we've broken away completely (version-wise) we can change a lot of stuff now - but do need to keep compatibility with existing scripts in mind - so while the main UI might move to new scheme, all the old constants are still needed. 

So I think this is both a bug (as in the constant is missing and needed) and a documentation issue (hey, did you know there are now these colour constants as of 2.5).

raphaelcoeffic, 2021-08-02:
By the way, @JimB40 was suggesting he could provide a mapping table that we could use to provide backward compat to 2.3. This should however be considered "deprecated" and might be gone in 1-2 years (or even earlier if we have a consensus).

lshems, 2021-08-02:
I think you need to define the strategy very clearly.

If you want scripts to run on opentx and edgetx, you need to have some kind of compatibility.

If you decide that that is not a goal, break the link now, instead of trying to maintain compatibility with something you won't support.

Personally, I will ditch opentx and will distribute edgeTX as part of my distro as soon as I find the time to do so.

If I want compatibility with opentx, o can do a loader script for the missing constants.

For me, the most important thing is alignment of BW and colour radio lua LCD functions, as I want to create only one (autoadjustable) script. The lcdtextsize function added by Jesper is what I had already done in lua by adding my own tables. But that is the approach to create a true LUA script ecosystem.

Sizeable scripts run wherever you like. One time development, one time layout

raphaelcoeffic, 2021-08-02:
Very good questions for which we need community input! Ditching compat is really easier!

> Personally, I will ditch opentx and will distribute edgeTX as part of my distro as soon as I find the time to do so.

:-)

> For me, the most important thing is alignment of BW and colour radio lua LCD functions, as I want to create only one (autoadjustable) script. The lcdtextsize function added by Jesper is what I had [...]

Yes, that would be nice, indeed! We did it with the BF LUA scripts at the time, but it was no fun.

lshems, 2021-08-02:
As for the color radios: are we going to stick to the palette/theme approach, or going to individual command colours. 

Perhaps move to a discussion? But it is relevant to these kind of changes.

pfeerick, 2021-08-03:
@lshems Maybe follow https://github.com/EdgeTX/edgetx/pull/540, as exactly this has been raised a possibility ;)

JimB40, 2021-08-04:
Let's move to #540. This be closed IMHO

closed 2021-08-04 by raphaelcoeffic

pfeerick, 2021-08-05:
I don't agree, from a compatibility standpoint. Does this mean the scope of #540 going to be broaden to include compatibility with LUA colour defines (in addition to as implementing the new colour mapping at the C++ level)? Or should this instead be rolled into a new issue, focused on colour but from a LUA standpoint? As I would think in dumping all the colour stuff *except* for the new colour constants, and making the old constants available only for LUA/widgets would be the way to go, if practical. 

In other words, TEXT_INVERTED_BGCOLOR is missing at the LUA level, and AFAIK that PR #540 does nothing to address this, and was not intended to do so. It is **_not_** valid at present to say that this doesn't need to be implemented for LUA (and LUA only), as we're not at a major version 'API breaking changes' version yet, and need to preserve as much of the expected behavior as possible.

jfrickmann, 2021-08-05:
The idea of #540 is to do a system-wide overhaul of the colors. Replace the existing 36 palette colors with the 8 new colors defined by @JimB40. For Lua, we will move from a palette based color definition to RGB, and provide a Lua function to get the current RGB value of a palette color.
I don't think that we can make a solution that would be 100% compatible with the old stuff, but I am of course open for suggestions. If we provide the old colors as RGB values, then lcd.setColor(area, color) will not work, and if we provide them as palette colors, then they will not work with the new RGB based color flags, but have to be converted from palette index to RGB before using.

pfeerick, 2021-08-05:
At the end of the day, if requiring the end user to use EdgeTX 2.4, and not be able to use a newer version in order to have any older lua scripts that used the existing definitions is the end product here, IMO that is not acceptable, as this has not been done in a major version change, which should be the only time _breaking_ APIs is allowed. 

Look at it from the perspective from the end user... they want to use EdgeTX 2.5/2.6 because of X, Y, Z great new features, but because of this break in behaviour, their X or Y lua script that they use all the time now has unreadable colours, so they throw their hands up and go back to OTX. We can't automatically assume they will know how to fix it, especially if we want to attract more people to use ETX. 

Perhaps I'm being somewhat naive/myopic though - as is there actually a significant body of LUA scripts that use the pre-defined colour palette values instead of defining their own colours? As if there isn't, this is a relatively pointless soapbox moment! 😆 If there are only a handful, maybe that can be fixed via PRs and/or documentation on how to fix any that break in that regard post 2.5.

raphaelcoeffic, 2021-08-05:
For 2.4, the compatibility was not perfect either, but we had “some compatibility”. Be it because the same constant were defined. So let’s define these constants in LUA only. Whether these constants represent the color itself or an index depends on the API. The goal should be to have the scripts which were provided by default (mostly FrSky and MPM stuff) to work in an acceptable way (does not need to be pixel exact). Otherwise we will have to provide new versions of all these scripts, which will produce additional work. As for the “big” scripts, they provide their own colors via setColor() anyway.