EdgeTX/edgetx #276: double conversions

lshems, 2021-06-19:
When opening SIMULATOR for T16 with FILE option, and choosing OpenTX2.3.12 file, it gets converted correctly, including model names and input line names.

When closing the sim, and restarting, it will do the conversion for a second time, messing all up.

raphaelcoeffic, 2021-06-20:
Now this is funny! Thx!

raphaelcoeffic, 2021-06-20:
@lshems which simulator did you mean? I just tried with `simu`, and it works, as far as I can see. No second version. Is this with the simulator using Qt?

lshems, 2021-06-20:
It's is with the simulator that is started with ./simulator24

lshems, 2021-06-20:
Also, the message is a little bit vague before conversion, and you can't prevent it from starting. It says something like 'sdcard needs conversion' which is not what is going to happen. The models are converted Inthe otx file.

raphaelcoeffic, 2021-06-20:
> Also, the message is a little bit vague before conversion, and you can't prevent it from starting. It says something like 'sdcard needs conversion' which is not what is going to happen. The models are converted Inthe otx file.

Ok, then this is a companion error, it is probably still using wrong directory names, or something like that. @rotorman could you please have a look at it?

lshems, 2021-06-20:
I will post a screenshot later. Still having some coffee first.

To reproduce, take X9D+.otx from the closed issue on 'some Lua model functions...' put it as file to load on the simulator profile, and start the simulator. It will convert all correct. Close the simulator and repeat

rotorman, 2021-06-20:
@lshems From the title, you are likely talking about #227. Cannot find any *.otx file there to test with?
Otherwise I have simulator24 running in Qt Creator under Ubuntu in Debug mode, so could try to figure it out.

rotorman, 2021-06-20:
I guess you mean this dialog here (tested with X9D+ simulator24):
[image]
I see this stems from
https://github.com/EdgeTX/edgetx/blob/3472bbdc781a93ef87fc6273b255889da921860e/companion/src/simulation/simulatorwidget.cpp#L269-L274
need to figure out why ret is 0, it does not hit the previous else case though:
https://github.com/EdgeTX/edgetx/blob/3472bbdc781a93ef87fc6273b255889da921860e/companion/src/simulation/simulatorwidget.cpp#L264-L267

lshems, 2021-06-20:
https://github.com/EdgeTX/edgetx/files/6680426/t16_23.zip

Load the Ubuntu simulator with the profile dialog, choose this file as file to load.

Start, conversion begins after an SD card message, close, and load again in the same way

lshems, 2021-06-20:
Change .zip to .otx of course

rotorman, 2021-06-20:
Have Jumper T16 now running in debug session in Qt Creator. Started with:
[image]

The following comes out of debug output:
[DebugOutput.txt](https://github.com/EdgeTX/edgetx/files/6682418/DebugOutput.txt)

lshems, 2021-06-20:
Ok,but it starts the conversion correct no?. Just SI and SJ warnings, but the conversions seem ok.

lshems, 2021-06-20:
And then, after closing, it tries to convert the (now already converted) file again

rotorman, 2021-06-20:
Yes, I see that it tries to convert every time one launches the simulator.
[image]

[image]

I see also that before this happens, the execution in the Qt simu app is passed to sim library at:
https://github.com/EdgeTX/edgetx/blob/3472bbdc781a93ef87fc6273b255889da921860e/companion/src/simulator.cpp#L352
I wonder if it happens, as there is no way for the radio simulator library (not the simulator binary) to write back the converted otx file into the host filesystem.
Did this work with OTX 2.3?

rotorman, 2021-06-20:
Interestingly the otx file that got converted by the simulator differs in the content, except the models.txt. All binary files have been touched. You can compare here with the original file from above:
[t16_23_converted.zip](https://github.com/EdgeTX/edgetx/files/6682548/t16_23_converted.zip)
Thus, the question is, why is a new conversion triggered. This should be irrespective of the simulator, and run similarly on physical hw/firmware.

rotorman, 2021-06-20:
By looking at the converted radio.bin, one sees that the tag is still old (0xDB = 219):
[image]

Also the models have the old tag, here exemplary model1.zip:
[image]

Seems that Companion24/Simulator24 do not cleanly save the new format.

@elecpower would you have tips, or be able to help out? Thanks!

elecpower, 2021-06-20:
From memory the Companion Simulator makes a copy of the model and settings to be simulated and any changes are discarded when the simulator is closed. The simu saves it changes.

rotorman, 2021-06-20:
@lshems please test PR #286 - for me this solves the issue.

lshems, 2021-06-20:
Ok, so help me a little, just getting around here. 

I have created an ubuntu according to the instructions. All fine. I can pull the head branch, also fine.
But how do I pull your PR?

rotorman, 2021-06-21:
@lshems `clone --recursive -b SimuCompFolders https://github.com/rotorman/edgetx.git pr286test` will create a local folder pr286test for you.

Please keep in mind, as @raphaelcoeffic hinted at #286, although this fixes the issue with repeated conversions, it might go amok with Companion, as the new data structure of version 220 needs to be additionally added there. Currently Companion is aware of data structures up to 219 only. This means, to get also Companion working, more work is required.

lshems, 2021-06-21:
I don't need companion,(yet).

A working simulator starting on an otx file would do for the moment. So without the reconversions.

Still a little puzzled on how that should work on a real radio, when doing the conversion.

raphaelcoeffic, 2021-06-21:
> I don't need companion,(yet).
> 
> A working simulator starting on an otx file would do for the moment. So without the reconversions.
> 
> Still a little puzzled on how that should work on a real radio, when doing the conversion.

I see, but the simulator you're using is heavily based on companion, which is the issue at hand right now.

lshems, 2021-06-21:
Just for my understanding, the simulator is depending on companion for defining the 'load settings' for the profile, or for much more?

Because I would reckon the simulator is more sort of a hardware emulator. Otherwise you would have to update all things on the firmware as well as for the simulator??

Just trying to graps the architecture to be able to understand dependancies.

raphaelcoeffic, 2021-06-21:
> Just for my understanding, the simulator is depending on companion for defining the 'load settings' for the profile, or for much more?
>
> Because I would reckon the simulator is more sort of a [...]

It does some reading of the model/radio file to be able to setup the controls. Then it seems it tweaks  the files and repacks them (did not know).

lshems, 2021-06-21:
I always had the feeling that opentx was spaghetti code, based on the magic numbers mayhem in Lua API.

This is another indication. I'll better keep far from it....:)

raphaelcoeffic, 2021-06-21:
> I always had the feeling that opentx was spaghetti code, based on the magic numbers mayhem in Lua API.
> 
> This is another indication. I'll better keep far from it....:)

Let's say: it has a long history.

pfeerick, 2021-06-23:
> I always had the feeling that opentx was spaghetti code, based on the magic numbers mayhem in Lua API.

Magic numbers? Are you implying there is magic involved in OTX? :-P

lshems, 2021-06-23:
Well, it runs by chance, so ...

raphaelcoeffic, 2021-11-22:
@lshems I believe this issue is fixed, right? Can you please confirm?

lshems, 2021-11-22:
I can correctly read a otx file and it gets converted in the simulator.

rotorman, 2021-11-22:
@lshems meaning, we can close the issue, correct?

lshems, 2021-11-22:
Yes :)