EdgeTX/edgetx #278: Anounce Model Name when switching models

jrwieland, 2021-06-20:
Upon startup the model name is announced (assuming you have a name.wav for each model)

When switching models the name is is not announced.

This is most likely since the screen stays on the selection screen (when the model information is loaded into memory).
Upon return to the main screen (via return) the verbal clue I was used to when switching models is no longer present.

This may be by design (staying on the selection screen) if so....
is it possible to have the name.wav announce when a model is selected from the selection screen.

pfeerick, 2021-06-21:
In principle, I don't see why not... 

You could probably do this right now with a special function, as the special functions I have for for switch positions read out as soon as the model is selected (unless programmed not to). But it certainly isn't as nice as having an option/feature in the model to speak the name when loaded.

philmoz, 2023-10-05:
This seems to be fixed in 2.8 and later - can probably be closed.

pfeerick, 2023-10-05:
Thanks for the reminder... I seem to remember fixing this during re-writing some of the model select code... it wasn't using the same functions as during startup.

closed 2023-10-05 by pfeerick