EdgeTX/edgetx #280: Calibration/Jitter Issues?

sprrrry, 2021-06-20:
I'm having some weird issues with the calibration/jitter on the gimbals on my tx, have been noticing it for a while but not sure what it actually is. There is very minimal jitter(1499-1501us) when I calibrate it which is normal for jumper transmitters but when I reboot it, it is like some other calibration value is being loaded sometimes. The sticks are jittering around 1496 or 1504 which should be around 1500 but then after rebooting, sometimes at randomly gets fine.  It's fine on OpenTx (around 1499-1501) even after rebooting several times, looks like some bug in EdgeTx.
When it is fine (jittering around 1500) after rebooting, it stays that way for as long the transmitter is on and when it is jittering around 1496 or 1504 it stays like that. Also ADC Filter is switched on.

raphaelcoeffic, 2021-06-20:
This will be a rough one, as I have 0 clue what that could be...

lshems, 2021-06-20:
Can you check with ADC filtering on and off? It might be the filtering.

sprrrry, 2021-06-20:
The problem remains whether ADC Filter is on or off but with the ADC filtering off the jitter is much worse/violent(same as ADC Filter off in Opentx). I don't think filtering is the problem because it remains fine all the time till it's powered on if it was fine on startup.

lshems, 2021-06-20:
Ok. Thanks for testing.just trying to help to exclude things. This was one!

sprrrry, 2021-06-23:
I have attached the video of the issue. In the video it powers on thrice with wrong calibration and then correctly but this is completely random. 

https://user-images.githubusercontent.com/38971821/123093232-12eaa200-d449-11eb-8850-a2fc7348156e.mp4

raphaelcoeffic, 2021-06-23:
@sprrrry do you notice a difference in the values shown in “analog” diagnostics? Is it different, or exactly the same?

sprrrry, 2021-06-23:
It is different, there is some deflection there too.

TramALot, 2021-06-23:
T18s RadioKing (jumper t18) adc filter off, with it on mine flys different/
badly. I use elrs with serial based rx & crsf protocol

raphaelcoeffic, 2021-06-23:
> It is different, there is some deflection there too.

Ok, these values are directly what we get from the ADC, so it tends to show there is something with the hardware, I’m afraid.

sprrrry, 2021-06-23:
> > It is different, there is some deflection there too.
> 
> Ok, these values are directly what we get from the ADC, so it tends to show there is something with the hardware, I’m afraid.

I also thought the same but I just did a calibration and intentionally put the sticks wrong for the center calibration and then in 'analogs' when the stick was 'actually' centered(not centered in software) the analog reading was also quite highly deflected from zero, so I think the analogs reading is dependent on the calibration and not the hardware ADC value.
Moreover I think the hardware is fine because I'm not having this problem in OpenTx.

raphaelcoeffic, 2021-06-23:
> > > It is different, there is some deflection there too.
> > 
> > 
> > Ok, these values are directly what we get from the ADC, so it tends to show there is something with the hardware, I’m afraid.
> 
> I also thought the same but I just did a calibration and intentionally put the sticks wrong for the center calibration and then in 'analogs' when the stick was 'actually' centered(not centered in software) the analog reading was also quite highly deflected from zero, so I think the analogs reading is dependent on the calibration and not the hardware ADC value.
> Moreover I think the hardware is fine because I'm not having this problem in OpenTx. Edit - Also it stays at the calibration value(correct or incorrect) for as long as the radio is on(doesn't necessarily change after a reboot).

Aha! Did you look at the HEX values (left), or the translated values (right)? The right values will always be the same, as it does use the calibration. The left column with Hexadecimal numbers does not.

sprrrry, 2021-06-23:
Yes I was checking the wrong values(translated) :), the hex values are also different though. But I find this really weird because when the outputs are 'off calibration', I have to push the throttle really hard downwards to reach the hex value which gives 1000us(hex value for 1000us is the same for the correct and incorrect tx session). Its 4444 or 444B at the same position physically. And the fact that I haven't experienced this in OpenTx makes me even more confused. I wonder if anyone else with a tx18s/t18 is having this problem

raphaelcoeffic, 2021-06-23:
> @raphaelcoeffic Yes I was checking the wrong values(translated) :), the hex values are also different though. But I find this really weird because when the outputs are 'off calibration', I have to push the throttle really hard downwards to reach the hex value which gives 1000us(hex value for 1000us is the same for the correct and incorrect tx session). Its 4444 or 444B at the same position physically. And the fact that I haven't experienced this in OpenTx makes me even more confused. I wonder if anyone else with a tx18s/t18 is having this problem

I’m quite puzzled as well, really.

pfeerick, 2021-07-03:
@sprrrry This is for a Jumper T18 isn't it? Hall effect gimbals?

sprrrry, 2021-07-03:
@pfeerick This is the Radioking tx18s, same hall gimbals as the Jumper T18(non pro)

richardclli, 2021-08-03:
I think this issue should be closed, as a matter of facts, it is a hardware problem anyway.  Due to the poor design of the T18/TX18S hall gimbal.  The voltage output range of the hall sensors is about 0.6-1.4V (measured result).  And with Vref=3V, the ADC can only gives 4096 * (1.4-0.6)/3 = 1092 levels, and as a matter of fact usually the last digit is not that reliable, it can gives only 500 stable levels, and that is why there are jitters.

richardclli, 2021-08-03:
ADC filters cannot help in this case, anyway.  The filter just takes 4 samples with an average calculation, which will not help with the poor hardware design that will amplify the last digit noise of the ADC conversion.

richardclli, 2021-08-03:
So, nothing can be fixed in software side.

closed 2021-08-03 by raphaelcoeffic