EdgeTX/edgetx #281: Insufficient accuracy of GPS telemetry

Scratch1983, 2021-06-20:
Insufficient accuracy of displaying GPS coordinates. XX°XX' YY°YY' is a 1 km x 1 km square.

1) It is more convenient to use coordinates in decimal format XX.XXXXXX YY.YYYYYY
2) It is necessary to improve the display accuracy to at least 4 decimal places.
3) On the home screen widget, it would be better to display coordinates in 2 lines.

 - EdgeTX version 68986020
 - HardwareRM TX16S
 
[image]
[image]

lshems, 2021-06-20:
I need to make work of the GPS WIDGET. 

It would have an option to show coordinates in several formats, including miles/km/meters from pilot ref. That is true only thing that makes sense to show in my opinion. Why would you want to have the GPS coordinates readable in this earth circumference scale.

There will be a QR code to scan for portable apps.

lshems, 2021-06-20:
One format will also be distance and bearing

Scratch1983, 2021-06-20:
At the moment, the main problem is low display accuracy.

lshems, 2021-06-20:
If I could, I would seperate the Lat and Lon part and lake it seperate sensors. The entire idea of this sensor list is to show they are there and working, not to show the value per se. That is done in a widget

Scratch1983, 2021-06-20:
The widget also has low accuracy. If you are planning to make another widget, then ok.

lshems, 2021-06-20:
Well, that is the idea.

Scratch1983, 2021-06-22:
[image]
This is OpenTX. Do the same in EdgeTX, please.

pfeerick, 2021-06-23:
Perhaps that's a side effect of the GPS settings being removed from the radio setup menu? As one of the options was `Coordinate format` which allowed you to toggle between DMS and NMEA (the other two options being `Timezone` and `Adjust RTC`).

Scratch1983, 2021-06-23:
I don't think the problem is in the format.
You need to add the number of digits after the decimal point.

raphaelcoeffic, 2021-06-23:
> I don't think the problem is in the format.
> You need to add the number of digits after the decimal point.

This is just a minor issue, we have the data, so it’s only about displaying them.

raphaelcoeffic, 2021-06-23:
@Scratch1983 I re-added the "minutes" part of the GPS position. You should be able to see them now in the "Value" widget.

closed 2021-06-23 by raphaelcoeffic

ChrisOhara57, 2021-12-23:
Have you considered to use Open Location code as an alternative format for display.
An accuracy of 3.5 to 3.5m would only need 8 characters in the short version.
Nine character would already give an accuracy of 0.9 to 0.9 m.
And it's only one value instead of two.

pfeerick, 2021-12-24:
Probably not for this - but it would something good to do as a dedicated Lua GPS widget.

ChrisOhara57, 2021-12-24:
> Probably not for this - but it would something good to do as a dedicated Lua GPS widget.

A dedicated LUA Script does already exist.
https://github.com/opentx/opentx/issues/8539

pfeerick, 2021-12-24:
Sweet! Feel free to submit it as an addition to https://github.com/EdgeTX/edgetx-sdcard

ChrisOhara57, 2021-12-27:
> Sweet! Feel free to submit it as an addition to https://github.com/EdgeTX/edgetx-sdcard

Is this done by writing a pull request and attaching the script?

pfeerick, 2021-12-27:
Sort of, yes. If you're not familiar with github enough to do a PR, open an issue and attach the zip file there - and I'll do a PR to add it later. Remind me what transmitters it suits also.

ChrisOhara57, 2021-12-27:
> Sort of, yes. A PR is more a suggested change, so would actually have any new files or changes in place. If you're not familiar with github enough to do a PR, open an issue and attach the zip file there instead - and I'll do a PR to add it later. Remind me what transmitters it suits also.

Thank you, but I try myself if a pull equest is the right way.

commit 78db9bcd (2021-06-23): [color] Add minutes to GPS position display

Resolves #281