EdgeTX/edgetx #284: Lag navigating SD card

JimB40, 2021-06-20:
This I got from John (RCVideoReviews) so just copy-paste:

_I get some ridiculously bad lag navigating sd card.
I figured out on my own that a 4k allocation size seems to work best with otx. Anything larger causes stutter reading simple things like audio prompts.
I discovered this after formatting a 16gb card for otx and using the windows default allocation size of 8k. When I did that, "welcome to opentx" audio stuttered on load. Going back to 4k allocation size removed the stutter and that is repeatable.
It seems something under the hood may not be optimized for modern/larger cards.
small 256mb cards get a default allocation of 4k when being formatted in windows.
I had to use a non-windows format tool to get a 4k allocation size on a 16gb card._

Btw I'm experiencing 'Welcome to EdgeTX' stuttering too.

raphaelcoeffic, 2021-06-20:
> This I got from John (RCVideoReviews) so just copy-paste:
>
> _I get some ridiculously bad lag navigating sd card. I figured out on my own that a 4k allocation size seems to work best with otx. [...]

So it might be worth telling people how to format for now. I would need to research that, but I think the SD card driver has only 4KB buffer, which is already quite much for the X9D+. We could raises these buffers on color LCD targets however.

pfeerick, 2021-06-21:
On Windows, I think [Rufus](https://rufus.ie/) would work fine for that.

raphaelcoeffic, 2021-06-21:
Or “SD format”, which is publish by the SD consortium, which is required for compliance tests.

pfeerick, 2021-06-21:
Yeah. `SD Formatter` probably the best 'single' tool to recommend since it's both Windows and Mac available. I just found it rather pointless installing that just for SD cards when Rufus did the same job, was a lot smaller, and didn't even need to be installed. :grinning:

lshems, 2021-06-21:
Just for fun. Check how often this issue is handled on OpenTX. 10 issues on audio stutter only.

It is not new at all, and I would be surprised if this reformatting solves all issues. (pleasantly though).

closed 2025-05-16 by JimB40