EdgeTX/edgetx #287: ELV and THR  Trim Reverses

MervDale, 2021-06-21:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-06-21:
This one seems to keep coming back! :laughing: https://github.com/EdgeTX/edgetx/pull/45

raphaelcoeffic, 2021-06-22:
@MervDale I checked it on the T18, which is supposed to be the same (at least it uses the same target) and guess what? It works perfectly, trims are correct. So I would suggest you check the connections and possibly invert these 2.

raphaelcoeffic, 2021-06-22:
> @MervDale I'm not having any such issues on my TX18S on mode 2, what mode are you using? Maybe there is some bug in the mode you are using.

Maybe you can post a picture of the trims connection on yours so we can compare? Mine is ok as well.

pfeerick, 2021-06-23:
Check the `SYS -> Hardware -> Buttons` page... are the right trims updating there? If they are right there, the trims are right, and it's a UI problem. I had the same thing happening with #45... and right around then I put a new SD card in so radio and model settings got reset, and it came good... so I'm wondering if somehow something can get stuck in the radio settings?

raphaelcoeffic, 2021-06-23:
> Check the `SYS -> Hardware -> Keys` page... are the right trims updating there? If they are right there, the trims are right, and it's a UI problem. I had the same thing happening with #45... and right around then I put a new SD card in so radio and model settings got reset, and it came good... so I'm wondering if somehow something can get stuck in the radio settings?

I don't think this is an issue with the software. We have 2 ppl with the trims placed correctly on that radio. I believe this is more an issue with that particular radio. Pictures should help to see the problem.

pfeerick, 2021-06-23:
If the correct trims are updating in the hardware page (i.e T3 updates when you press T3), surely that means you don't have to open the radio, and that it's a glitch somewhere else? 😆

raphaelcoeffic, 2021-06-23:
> If the correct trims are updating in the hardware page (i.e T3 updates when you press T3), surely that means you don't have to open the radio, and that it's a glitch somewhere else (i.e. **_not_** a wiring/hardware issue)? Hence why I suggested checking. 😆 Next up would be the mode settings, as sprrrry suggested, as have the wrong stick mode will swap T3 and T2 (as well as the sticks just in case he hadn't noticed that 😁)

I don't think the stick mode swaps the trims. They stay in place, just have different meanings. From what I can see on my T18, everything is in place. This has been confirmed by others with T18 (/TX18S).

pfeerick, 2021-06-23:
Unfortunately, it *does*... so that probably explains what has been happening, and why the there has now been two or three PRs or changes that keep swapping the trims back and forward... 

I'm a mode 2 flyer, and T3 moves the trim on the left (throttle) side of the screen where it should be, and T2 is on the right (elevator) side of the screen where it should be. If I change to mode 1, T3 correctly alters the elevator, but now the trim on the right side of the screen moves instead of the one on the left. And the same is true for T2.

raphaelcoeffic, 2021-06-23:
> Unfortunately, it _does_... so that probably explains what has been happening, and why the there has now been two or three PRs or changes that keep swapping the trims back and forward...
> 
> I'm a mode 2 flyer, and T3 moves the trim on the left (throttle) side of the screen where it should be, and T2 is on the right (elevator) side of the screen where it should be. If I change to mode 1, T3 correctly alters the elevator, **_but_** the trim on the right side of the screen moves instead of the one on the left. And the same is true for T2.
> 
> And whilst I don't fly modes 3 and 4, it seems like they also get screwed up.

raphaelcoeffic, 2021-06-23:
> Unfortunately, it _does_... so that probably explains what has been happening, and why the there has now been two or three PRs or changes that keep swapping the trims back and forward...
> 
> I'm a mode 2 flyer, and T3 moves the trim on the left (throttle) side of the screen where it should be, and T2 is on the right (elevator) side of the screen where it should be. If I change to mode 1, T3 correctly alters the elevator, **_but_** the trim on the right side of the screen moves instead of the one on the left. And the same is true for T2.
> 
> And whilst I don't fly modes 3 and 4, it seems like they also get screwed up.
> 
> i.e. it's like the trim _displays_ are hard coded that the throttle stays on the left, rudder on the left, elevator on the right, aileron on the right, irrespective of where the sticks might be because of mode.

Oh! I see! Then I'll need to dig into this. (sorry for close/re-open, I hit the wrong button....)

pfeerick, 2021-06-23:
lol... yeah... it's a really curly one... hard to pin the sucker down... and it was probably because I edited my reply (again!) 😆

commit cdd6d18d (2021-06-23): Fixed trims display depending on stick mode

Resolves #287
closed 2021-06-23 by raphaelcoeffic