EdgeTX/edgetx #288: Tx16s: Potis missing in global functions (S1/S2)

Betriebsrat, 2021-06-21:
**Is your feature request related to a problem? Please describe.**
Tried to bind the volume to S1 or S2 on global functions. S1 and S2 are not selectable

**Describe the solution you'd like**
S1 and S2 added

raphaelcoeffic, 2021-06-21:
@Betriebsrat Can you please state which exact version you tested on please?

Betriebsrat, 2021-06-21:
Aww forgot the most important part sorry....
It was branch 2.4 (non-rc) via flasher tool.

Betriebsrat, 2021-06-21:
2.4.0 (78a0a457)
2021-06-20
18:15:08

pfeerick, 2021-06-21:
Just double checking this... 'value', not the 'switch' that you set the pots on... and I can still set them (as the value, not the switch) with RC2 on TX16S

[image]

Betriebsrat, 2021-06-21:
user error in my part sorry.

closed 2021-06-21 by Betriebsrat

pfeerick, 2021-06-21:
All good... I won't mention the number of times I've complained to Raphael about something and he said I was doing it wrong! :rofl: