EdgeTX/edgetx #290: TX16S: Lateral Cyclic Source changes when it's weight is adjusted.

ZZ-Cat, 2021-06-21:
**Description:**
Changing the Weight Value also changes the Lateral Cyclic Source.
The same thing also happens (but the other way around), when you change the Lateral Cyclic Source - IE The weight value changes with it.

**To Reproduce:**
In the Heli Setup tab of the Model Settings menu:
1.) Highlight the weight of the Lateral Cyclic Source.
2.) Click the scroll wheel, when the highlight turns green, increment this value by one.
3.) Click the scroll wheel again.
4.) Scroll up to the Lateral Cyc. Source & it will change, as soon as it's highlighted.
5.) Scroll back to the weight value & repeat steps 2 through 4, & you will see that it does this for every possible input option for the Lateral Cyclic Source.

For the other way around:
Simply change the Lateral Cyclic Source & you will immediately see that the weight number has changed. If you iterate through every possible input option for the Lateral Cyclic Source, the weight number will change according to what you have selected.

**Expected behavior:**
When the weight value is changed, the Lateral Cyclic Source should remain unchanged.
The same is true, when the Lateral Cyclic Source is changed - the weight should stay at the same value that was set previously.

**General information:**
 - Firmware: EdgeTX nightly 78a0a457.
 - Hardware: RadioMaster TX16S.

raphaelcoeffic, 2021-06-21:
@ZZ-Cat thx for reporting! I might have additional question later, as I have really no clue how a Hell setup works.

ZZ-Cat, 2021-06-21:
> @ZZ-Cat thx for reporting! I might have additional question later, as I have really no clue how a Hell setup works.

It's a good thing I fly helis. =^/,..,^=

raphaelcoeffic, 2021-06-21:
@ZZ-Cat ok, this was really easy :-) Fix has been pushed to 2.4.

closed 2021-06-21 by raphaelcoeffic

ZZ-Cat, 2021-06-21:
> @ZZ-Cat ok, this was really easy :-) Fix has been pushed to 2.4.

Sweet as. Thank you _so_ much. =^/,..,^=

commit 5e08f1c1 (2021-06-21): [color] Fixed lateral cyclic source/weight editing

Resolves #290