EdgeTX/edgetx #291: Baud Rate not selectable

Betriebsrat, 2021-06-21:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature.
You can add your comment there, or react so that we know know much interest there is in the feature request.*

**Is your feature request related to a problem? Please describe.**
The baud rate is no longer selectable via settings. Seems to be opentx removed that earlier releases stating that this is unneeded.
If you flash ELRS to an pre 2018 Frysky R9M without doing the resistance mod you need to select a lower baud rate. Further details: https://github.com/ExpressLRS/ExpressLRS/wiki/Inverter-Mod-for-R9M-2018

**Describe the solution you'd like**
Baud Rate being selectable again. Ultra low prio tho...

raphaelcoeffic, 2021-11-01:
@Betriebsrat I believe baudrate is now selectable on all platform in 2.5.0.

gagarinlg, 2022-07-20:
can this be closed?

pfeerick, 2022-07-21:
Yes. It's configurable on a per model basis for CRSF.

closed 2022-07-21 by pfeerick