EdgeTX/edgetx #293: Screenshot previews does not display correctly

NemesisXB, 2021-06-21:
**Describe the bug**
Screenshot previews does not display correctly. The image is correct, just not displayed correctly.

**To Reproduce**
Make a screenshot. Go to SYS > SD Card > Screenshots > view a screenshot

**Expected behavior**
Expecting the screenshot preview to look like the actual screenshot

**Screenshots**
Screenshot of the screenshot preview
[image]
The actual screenshot shown in the preview
[image]

**General information**
 - EdgeTX version (2.4.0-RC2r)
 - Hardware (RM TX16S)

raphaelcoeffic, 2021-06-21:
@NemesisXB probably another surprise for using the newest Stb lib.... I would love to drop BMPs as well, and use PNGs (if the compression code fits).

NemesisXB, 2021-06-21:
> @NemesisXB probably another surprise for using the newest Stb lib.... I would love to drop BMPs as well, and use PNGs (if the compression code fits).

It is obviously low priority. just thought I would create the issue so it is at least logged somewhere. And PNG would be awesome. I couldn't even load the screenshot directly onto gihub because github does not support BMP, had to convert it first.

pfeerick, 2021-06-22:
> I couldn't even load the screenshot directly onto gihub because github does not support BMP, had to convert it first.

Indeed... it's annoying that you can't drag and drop the files into an issue... Have to keep opening in gimp so I can copy 'n paste... :-/

And yes, I have the reddish screenshot previews also, just hadn't gotten around to lodging an issue.

LimitlessGreen, 2021-08-11:
I suggest imagemagic. It's an oneliner: `convert a.ext1 b.ext2`
If you really want to drop a BMP here, just zip it. :-)

closed 2022-09-18 by pfeerick