EdgeTX/edgetx #295: RADIO - GLOBAL VARIABLES bugs

JimB40, 2021-06-21:
1. GV name not visible when GV was at least once active
[image]

2. No "Override' function in Func list

3. After 'Clear' command issued from pop-up background and '---' still visible.
It goes away once 'Edit' and RTN or user goes to other tab then returns to Global Variables TAB
[image]

 
**General information**
 - EdgeTX version 2.4.0 RC2
  - TX16S

raphaelcoeffic, 2021-06-21:
> No "Override' function in Func list

I think this is intentional. The "Override" function has always been considered "harmful". But let's see, could be just a compilation option (I think it is).

JimB40, 2021-06-21:
I know tt is is compilation option 'nooverridech' 
But then if you compiled with this why there is 'Override' function in MODEL -> SPECIAL FUNCTIONS and there is no in RADIO -> GLOBAL FUNCTIONS

raphaelcoeffic, 2021-06-22:
> I know. It is is Companion/compilation option 'nooverridech'
> But then if you compiled with this why there is 'Override' function in MODEL -> SPECIAL FUNCTIONS and there is no in RADIO -> GLOBAL FUNCTIONS

Now I see. This is again intentional, because global functions are more related to "radio", and thus Override is allowed only model functions.

raphaelcoeffic, 2021-06-22:
@JimB40 and it seems to be the title color has already been fixed (I use 2.4 HEAD, and the title is visible, not like in your screenshot).

closed 2021-06-23 by raphaelcoeffic

JimB40, 2021-06-23:
testing 68cb9c2
still exist.
1. GV name disappear in edit mode if GV was active then deactivated (test with Toggle switch)
3. Same empty leftover after Clear command.

raphaelcoeffic, 2021-06-23:
> testing [68cb9c2](https://github.com/EdgeTX/edgetx/commit/68cb9c2d92cb6bfefb79dd8f2395a0e6b0562006)
> still exist.
> 
> 1. GV name disappear in edit mode if GV was active then deactivated (test with Toggle switch)
> 2. Same empty leftover after Clear command.

Can we make a new one, that was quite a package.

commit 68cb9c2d (2021-06-23): [color] Patch unavailabel global/special function if not available

Resolves #295