EdgeTX/edgetx #297: RADIO HARDWARE missing TopBar text

JimB40, 2021-06-21:
Goto RADIO -> HARDWARE

1. Press Analogs
[image]

Should display:
HARDWARE
Analog inputs test

2. Press Keys
[image]

Should display:
HARDWARE
Keys & switches test

**General information**
- EdgeTX version 2.4.0 RC2
- TX16S

raphaelcoeffic, 2021-06-21:
At least someone is testing 🤓

eshifri, 2021-06-21:
BTW, PGUP and PGDN sensors are swapped: when PGUP is pressed PGDN is set to 1.

raphaelcoeffic, 2021-06-22:
Fixed by #305

closed 2021-06-22 by raphaelcoeffic

raphaelcoeffic, 2021-06-22:
> BTW, PGUP and PGDN sensors are swapped: when PGUP is pressed PGDN is set to 1.

I think this is correct, I just checked. It is however sometime a bit mind-boggling. To the right is considered "Down", which is why "PGDN" is "Next Page" (I think, at least).