EdgeTX/edgetx #298: RADIO VERSION keys stuck

JimB40, 2021-06-21:
1. Go to RADIO VERSION
2. Touch the screen

PAGE>   PAGE<   &  EXIT doesn't work anymore

**General information**
- EdgeTX version 2.4.0 RC2
- TX16S

raphaelcoeffic, 2021-06-21:
Oh! 😳

eshifri, 2021-06-21:
I cannot reproduce it: all buttons are working for me.

raphaelcoeffic, 2021-06-22:
> I cannot reproduce it: all buttons are working for me.

Neither can I.

pfeerick, 2021-06-22:
Sorry, I can't repo it either on RC2.

raphaelcoeffic, 2021-06-23:
> Sorry, I can't repo it either on RC2. If you can't pin it down as consistently happening, or perhaps being related to touching a particular part of the screen implying there's a focus issue, this issue can probably be closed for now.

I can reproduce it. I’ll check what it does actually to the focus.

pfeerick, 2021-09-15:
I've been able to repo this also - seems more prevalent on the Radio => Version screen, but I have also been able to repo it once on the hardware tab also. Seemed to be something to do with touching around the "VERSION" title bar, so maybe there's like a one pixel focus gap? 😆

pfeerick, 2021-11-22:
I believe I found the trigger method for this... it's if you slide or swipe in the top bar.

gagarinlg, 2021-11-27:
I can confirm this. It is also the same problem for all windows that use tabs-

eshifri, 2021-11-27:
Interestingly enough touch buttons are still working.

gagarinlg, 2021-11-27:
> Interestingly enough touch buttons are still working.

The slide event is not handled by the top bar. So what happens is that the degault window slide handler takes over and sets the focus to the top bar part of the tabs window. But the top bar does not handle the input key events, so the keys are essentially dead, until you press on an icon without sliding. Then everything is back to normal.

PR #1180 (2021-11-29): 298 keys stuck
Fixes #298 

Summary of changes:
handle slide events on the tab bar and ignore touch end event with old coordinates when touch ends.
Please check on NV14
commit 15233e1a (2021-11-29): Fix sliding on inactive part of tab bar grabbing input focus (#1180)

* fix #298, keys not working, after sliding on tabs bar

* fix for touch end event with start x,y coordinate is triggered after the last slide event

* remove selecting tabs on the tab top bar by sliding as it does break sliding the tab bar.
closed 2021-11-29 by pfeerick