EdgeTX/edgetx #299: MODEL -> SETUP can't select proper channel range

JimB40, 2021-06-21:
1. Go to MODEL -> SETUP

2. Select as External RF Mode: GHST 
2a. Select end channel in Channel Range lower then 8th
2b. Try to select 8th again.
Stuck at 7th as max channel

3. Select as External RF Mode: SBUS 
3a. Select end channel in Channel Range lower then 8th
3b. Try to select 8th again.
Stuck at 7th as max channel

4. Select as External RF Mode: CRSF 
End channel in Channel Range can't be selected
It keeps channel range selected in other mode.

**General information**
- EdgeTX version 2.4.0 RC2
- TX16S

raphaelcoeffic, 2021-06-21:
> 1. Go to MODEL -> SETUP
> 2. Select as External RF Mode: GHST
>    2a. Select end channel in Channel Range lower then 8th
>    2b. Try to select 8th again.
>    Stuck at 7th as max channel
> 3. Select as [...]

I think they just fixed something like that in OTX. Should be easy to fix though.

phileville, 2021-07-26:
Also, with the internal MPM, you cannot select a channel above 12.

nm17, 2021-09-17:
@pfeerick @phileville can confirm this still happens on nightly, please look into it.

pfeerick, 2021-09-17:
I can also confirm for CRSF (and GHOST IIRC) it's cosmetic as you can't change it from the model setup. But thank you for confirming it's still there.

PR #755 (2021-09-18): Fix max # channels and default channel range
Resolves #299
commit 9b63f75e (2021-09-18): Fix max # channels and default channel range

Resolves #299
closed 2021-09-18 by raphaelcoeffic