EdgeTX/edgetx #300: MODEL -> INPUTS bugs

JimB40, 2021-06-21:
1. Empty line in pop-up
- Go to empty line & press ENTER

[image]

2. Delete item in pop-up should be named 'Clear'
Let's stay constistent across OS
- Clear wipes out data but not the line
- Deleter wipes out data and moves lines below up 1 line

3. After Paste or Delete command focus marker goes to first input
- Should stay on Pasted line or on nearest to do deleted line

**General information**
- EdgeTX version 2.4.0 RC2
- TX16S

JimB40, 2021-11-11:
General assumption is to reduce clicks/touches amount
So if only option is available in contextual menu we should consider executing this option instead of opening list with 1 item.

pfeerick, 2022-03-10:
Just some notes:

For 1, this appears to be the case system wide, so it looks like the minimum "menu" size is two rows... so that probably needs to be fixed back in libopenui, so may implicitly get fixed as part of LVGL work...

For 2, while I agree, I'm not sure this would be possible to do at this time, as the same menu is shown regardless of if it is a single input, or a multi-line input, and in the case of multi-line, it is actually Delete, but when there is only one entry, it would essentially be Clear. I don't think the menu supports dynamically changing the text (yet?). Maybe the menu lines could be cleared and re-added if there is only one input line left?

closed 2022-09-09 by pfeerick