EdgeTX/edgetx #301: Copy-Paste buffer not cleared - EM !!!

JimB40, 2021-06-21:
Pouwer up RADIO
1. Go to MODEL -> MIXES
2. Select line with data
3. Copy
4. Go to MODEL -> INPUTS
5. Press ENTER (Paste is shown even no input was copied)
6. Paste
Now different things happen including EM

**General information**
- EdgeTX version 2.4.0 RC2
- TX16S

closed 2021-06-22 by raphaelcoeffic

JimB40, 2021-06-23:
Still active 68cb9c2

raphaelcoeffic, 2021-06-23:
> Still active [68cb9c2](https://github.com/EdgeTX/edgetx/commit/68cb9c2d92cb6bfefb79dd8f2395a0e6b0562006)

Between Mixers and Inputs? Really? I tried any combination. Or did you find another?

pfeerick, 2021-06-23:
I think this is the wrong issue Jim ;)

JimB40, 2021-06-24:
It is :) need video for proof?
This time copied line in INPUTS and pasted in MIXES -> EM

pfeerick, 2021-06-24:
If you're still running RC2, yes, but if you run the latest nightly, which includes that commit it shouldn't. I can't reproduce it, as the paste option disappears as soon as you change from INPUT -> MIXES tab with either touch or page buttons.

JimB40, 2021-06-24:
Okay took 5d9498e8 and it's gone.
I had no RC2 but the build from yesterday. Still thx for pointing
Then I change my test workflow to use only night builds. Otherwise it's waste of time for testing and issue submiting.

commit b1768afc (2021-06-22): [color] reset clipboard on page tab switch

Resolves #301