EdgeTX/edgetx #306: TX16S - Default model will not load

oryfoxer, 2021-06-22:
https://user-images.githubusercontent.com/6994101/122861335-e289f300-d2e4-11eb-8bb0-898021cd6632.MOV

The default model will be overwritten when creating a new model. 

Checked models.txt and confirm models exist.

When trying to use the new model2 for expressLRS - the radio will pull telemetry, but the quad will show failsafe in betaflight. 

This prevents me from using ExpressLRS and multiple models at the same time.

raphaelcoeffic, 2021-06-22:
> When trying to use the new model2 for expressLRS - the radio will pull telemetry, but the quad will show failsafe in betaflight.

Is that right after creation of that model? Without any changes from one model to the other, or anything like that?

raphaelcoeffic, 2021-06-22:
> The default model will be overwritten when creating a new model.

The thing we really need to sort out is: why is the "model select" page empty for you? Normally, this means that `models.txt`was either empty, not readable, or otherwise corrupted when start the radio. It could load `model1.bin` (or whatever the name of file was), because it is still saved as "last model" in the radio file. But the models list needs to be accurate, otherwise it's difficult.

oryfoxer, 2021-06-22:
[Models]
model1.bin

Was my models.txt at time of flashing.

gagarinlg, 2022-09-15:
Is this still a thing?

closed 2022-09-17 by gagarinlg