EdgeTX/edgetx #307: Display Real-time Value of Channel on OUTPUTs page

grepboy, 2021-06-22:
On current build of OpenTX 2.3.11, the OUTPUTs page on theTX16S displays real-time the value of the highlighted channel.  
See example picture, output value is circled in Yellow.

Please include this feature in EdgeTX.

[image]

jmil-dev, 2021-07-04:
I'd add to this that because the new interface screen uses % values rather than uSec there's no way to know what the output values are limited between

pfeerick, 2021-07-04:
Actually, there probably is, but it is still admittedly clunky. 
[image]

JimB40, 2021-11-11:
Let's bring it back.
It's real pain in the ass to setup properly Output not seeing actual micro secs value while changing outputs ends
Bottom Bar is right place for informations.

raphaelcoeffic, 2021-11-11:
This can be implemented easily by integrating the different `ChannelBar` classes here:
https://github.com/EdgeTX/edgetx/blob/493c90d792f7815e89608fb65aa2d426e06485f8/radio/src/gui/colorlcd/channel_bar.h#L37-L48

pfeerick, 2022-03-11:
I think #1664 does this, and better, as it shows multiple channels rather than just the highlighted one. An option to show just a highlighted channel's value may still get implemented in the future though.

closed 2022-03-11 by pfeerick