EdgeTX/edgetx #311: SD Card Navigation Folder Operations

jmxp69, 2021-06-22:
Current long press of files in sd card browser offers pop-up action menu with "Execute, Copy, Rename, Delete"
Current long press of folders in sd card browser enters folder.
Current short press of folder enters folder.

This is not consistent. Long press on folder should yield action menu with "Copy, Paste, Rename, Delete"

I realize it's browser, but the actions available are inconsistent.

JimB40, 2021-11-11:
There is no (Execute, Copy, Rename, Delete) for folder ATM.
To make UI consistent I propose to switch off contextual menu on ENTER long-press when selection is focused on folder type element

philmoz, 2023-01-18:
Looking at the code, the file browser is based on the TableField class from the libopenui sub project.

There is no long press handler per-se. A long press is treated just like a short press.

The first time a file is clicked it is selected and if it can be previewed a preview is displayed.
When the currently selected file is clicked the popup menu opens.

To address this would require changes to the TableField class in libopenui, or replacing it with something else.

PR #5388 (2024-08-07): feat(color): add long press menu for directories in SD browser
Fixes #311

[image]

[image]
closed 2024-08-07 by pfeerick