EdgeTX/edgetx #319: update LUA getVersion() function

JimB40, 2021-06-23:
Add OS name in LUA getVersion() return values, as there is now no way to check what OS is running
"OS name" return value should be added as lastto preserve compatibility with existing LUA scripts.

Propsed LUA API change:

**getVersion()**

**Parameters**
none

**Return values** 
- multiple returns 6 values:
  - (string) OS version (ie "2.4.0")
  - (string) radio type: x12s, x10, x9e, x9d+, x9d or x7. (If running in simulator the "-simu" is added)
  - (number) major version (ie 2 if version 2.4.0)
  - (number) minor version (ie 4 if version 2.4.0)
  - (number) revision number (ie 0 if version 2.4.0)
  - (string) OS name (ie "EdgeTX" or nil in case of OpenTX)

pfeerick, 2021-06-24:
Just to be clear, the proposed change is that the 6th value would be returned as well, as the current API as of OTX 2.1.7 returns the first five values.

JimB40, 2021-06-24:
Correct. 
have we any luadoc in for edgeTX?
Like rhis one possibly
https://doc.open-tx.org/opentx-2-3-lua-reference-guide/

pfeerick, 2021-06-24:
Not yet. Coming soon:tm: ... I think UI and documentation will be the major focus of 2.4.1.

JimB40, 2021-06-24:
So we should have at least change log for this lua API atm. Issue? Thread?

pfeerick, 2021-06-24:
I dunno... the whole LUA API is in a state of flux AFAIK from the 2.3 => 2.4, so I think it will have to be from scratch, working through each function at a time to see what is changed, what hasn't, etc. I don't think there's much point doing a 'changelog' as such (yet) as it would be a dogs breakfast of a thing to understand. Especially if the API is still a moving target ;)

PR #323 (2021-06-26): Extend lua getVersion()
Resolves #319 

Extends lua getVersion() so that it returns a sixth parameter, the OS name, allowing scripts to identify if running on OTX or ETX. 

Didn't see a readily usable pre-existing define that had `EdgeTX` in it so added `VERSION_OSNAME`
closed 2021-06-26 by raphaelcoeffic