EdgeTX/edgetx #322: Global variables bugs

JimB40, 2021-06-23:
1. GV name disappear in edit mode if GV was active then deactivated (test with Toggle switch)
2. Empty leftover after Clear command
[image]

**General information**
 - EdgeTX 68cb9c2
 - RMTX16S

pfeerick, 2021-06-23:
GV (Global Value) or GF (Global Function)? If it's the later, I'm still on the latest nightly (so doesn't include that commit, but I don't think that matters) , and can reproduce that - i.e. GF3 for me is for screenshot of SHdn and I initially have GF3 in white in the header, yellow while active, and then it vanishes. 

Not sure there's anything wrong with that second point... clear is supposed to clear, not delete. If you delete instead of clear, it removes the row. If you clear, it clears the values, but leaves the row. What is a bug though is (at the global functions screen) if you clear instead of delete (as you have the choice of both initially), the context menu that appears for that row after clearing  only has Clear as an option. i.e. you've lost the option for delete, which in this case is the only valid option since it has already been cleared.

JimB40, 2021-06-24:
After ‘clear’ GF3 line should look like GF4 or GF5 on screenshot. Values are cleared It’s just UI display bug. If you change tab and go back it look correctly

pfeerick, 2021-06-24:
Latest nightly, no. 1 is till present. 

Ah, I'm with you now... to resolve no. 2, instead of this when clearing... 

[image]

... you want it to look like this... 

[image]

... which is consistent which how it looks if you revisit the screen later. Yeah, oopsie!

pfeerick, 2021-06-24:
#325 Addresses point 1 of this issue, so just point 2 needs doing ;)

Jhawk736, 2022-02-24:
Hardware is Jumper T16 using EdgeTx 2.6
Radio Global functions don't work.
Instead, special function which refers to each model act as global function.
Was setting S2 as volume control, its works via special function.
Is global function now disabled in the version ?

pfeerick, 2022-02-24:
No, global function for volume (and brightness) still work - perhaps you have not configured or enabled the global function - It should be configured like so... make sure the swith is set to ON (unless you use a switch to enable and disable pot volume control) and the function is enabled. 

[image]

Jhawk736, 2022-02-25:
Yes, exactly the same settings. I notice that I cannot enable it. When clicking the enable option, the GF1 letters should turn yellow.
Mine does not, but Special functions do turn yellow when enabling.
Any idea? Fw is edgetx-t16, thanks.

Jhawk736, 2022-02-25:
Further digging, found that Global function works in the 2nd model. 
Going back to the 1st model, Global function settings remains, but the enable button still don't work.
I have to keep screen and volume in Special function in order for Global function to work in other models.
It's weird but it's good to go.

commit 2f3919c4 (2021-11-02): Rebuild window when SF entry cleared
Resolves #322