EdgeTX/edgetx #326: Touch UI back button does tot function when editing Top Widgets

phileville, 2021-06-24:
**Describe the bug**
When editing the top bar widgets, the touch back button does not work. The only way to get out of the edit mode it is to use the hardware rtn button. This give the users the false impression that the touch screen is locked up, since you cannot swipe the main screen as well when in this mode.   

**To Reproduce**
Go to screen settings, select the TOP Bar setup widgets, notice that you cannot use theTouch UI Back button.

**Expected behavior**
Tough UI back button should work in this mode as well

**Screenshots**
Not needed

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-06-24:
Also when editing the panels, not just the top bar... ;)

gagarinlg, 2022-07-20:
This can be closed, right?

pfeerick, 2022-07-21:
Yes, I think this was fixed back in 2.7, but the fix not linked. Regardless, I can't repo either case this on 2.7.1 or `main` anymore.

closed 2022-07-21 by pfeerick