EdgeTX/edgetx #328: Horus Battery Percentage Widget not visible

3DRacer, 2021-06-24:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

JimB40, 2021-06-24:
This is external LUA widget.
I suggest to contact author to make adjustments for EdgeTX

lshems, 2021-06-25:
He is really responsive. Should be good to get him onboard for these kind of widgets.

3DRacer, 2021-06-25:
I have since followed the advice and wrote to the author about a solution. Now it remains to be seen

rotorman, 2021-06-25:
I had a quick look at the LUA widget at https://hobby4life.nl/?p=86
It is missing a fallback case in the ScreenUpdate() function if LiveVolt > 0.
To fix this just comment out the last elseif and change it to else.

Thus find the following line (if you copy pasted it from the website, then it should be line 240):
`elseif LocalContext.zone.w == 160 and LocalContext.zone.h == 32 then`
and change to just
`else`
That should make it to display the content again.

pfeerick, 2021-06-25:
> It is missing a fallback case in the _ScreenUpdate()_ function if _LiveVolt > 0_.
> To fix this, just change the last _elseif_ to _else_.

Thanks for that @rotorman At least it's not a bug introduced by ETX ;)

closed 2021-06-25 by pfeerick