EdgeTX/edgetx #330: Create Lua mixer script for Heli cyclic mixing

lshems, 2021-06-25:
When moving away from a special Heli page, a Lua mixer is proposed as alternative.

Apart from the other issues about reliability,this issue to create the lua mixer script.

You can assign it to me, as it is Lua.

I just have to now where it needs to go. Sdcard ?

JimB40, 2021-06-25:
Issue about reliability seem to be key one. So far it seems that there is no will  among heli pilots to fly 1500USD FB heli if LUA script (that can be killed by OS) is involved.

lshems, 2021-06-25:
That's to short. There are very few Heli pilots that need cyclic mixes. Those who do need those mixes, are not listed, so it's tricky to make such statements.

I do agree that it will be very likely, and not advisable to use scripts for the cyclic mixes NOW.

Bit ok, move this to another place, and open another issue to get the then createdclia script ported to an internal mix to avoid the use of scripts for in flight mixers.

pfeerick, 2021-06-25:
Reliability is the key question, meaning until some thought and discussion has gone into https://github.com/EdgeTX/edgetx/issues/329, for instance, this is one of those 'would be nice, but totally unreliable' ideas. As even if you made it, no-one would use it. 

I honestly think there are three main options:
1. keep the heli page, and made it so lua can access elements of it (this has also been proposed over on OTX, and they are willing to look if if a PR is done and are not against the idea in principal)
2. ditch the heli page, and use multiple mixer lines, moving the configuration to lua instead. 
3. do nothing

So maybe make that two options, and I'm leaning towards 2. I doubt the heli guys would be against it if there was a wizard script, and a configuration script bringing back at minimum the same functionality they have in the current heli page, whilst also opening the door for more. 

My reasoning? The more 'specialised' stuff we can take out of the main firmware the better. i.e. I don't use heli, and it would be nice if I don't have to rebuild the firmware to remove it (or if something changes, down the path, it's a matter of copying a folder onto the SD card, or to be honest, it's a couple of K of my 16GB SD card... I don't really care if it's there or not. But it's now no longer wasting my valuable transmitter firmware space, and I can go it and customise if if I wanted to.

lshems, 2021-06-28:
Don't know where I should publish it, and how. But here it is, a fully functional complete replacement (except for the rare X120 swash head, but it can be added if really necessary) of the heli mixer page using the standard LUA mixer script functionality.

Including variable angles for 1°-89° X heads,  variable angels for 91°-134° equal distance pivot heads, and variable angles for 135°-179° equal longitudinal length pivot heads.

Correct implementation of 135° and 140° heads (I think so at least ;))

Bonus: Expo on Throtthe input. (perhaps completely useless, LOL again).

Of course I would prefer this mixer to be created in C and be called in the normal mixer cycle. The CONCEPT of seperate mixers with multiple inputs and outputs for mixer parameters AND sources as is present for LUA scripts is really genius, and should in my opinion be maintained. 

Why not use only the definition part of the LUA mixer list, and have people choose between C mixers in the firmware (this HELI mixer should go there in my opinion) and "traditional" lua mixers.

Use the OpenTX 2.3.13 Taranis X9D+ OTX file to check how it works. Sources created are SRV1, SRV2, SERV3 and SRV5. They are put pair to pair on channel 23-28 against the CYC1, CYC2 and CYC3.

[HELI.zip](https://github.com/EdgeTX/edgetx/files/6729497/HELI.zip)
[OTX.zip](https://github.com/EdgeTX/edgetx/files/6729564/OTX.zip)

pfeerick, 2021-10-02:
If you want to "publish" it, how about a PR to add it to https://github.com/EdgeTX/lua-scripts since that's a thing now... Then this issue can get closed as you've created what the issue was about ;)

ror1948, 2025-11-10:
I get error in heli.zip
"attempt to perform arithmetic on a null value
ver 2.10.6
when i first ran it it said "rate" global was missing but now it does not say that.