EdgeTX/edgetx #333: RTC bridge is not disabled, if "Check RTC voltage" is not enabled

rotorman, 2021-06-25:
@Thx to top by Mike Blandford in RCGroups https://www.rcgroups.com/forums/showpost.php?p=47360301&postcount=82

If SYSTEM -> HARDWARE -> _Check RTC voltage_ is disabled, the VBatBridge is enabled during adcInit():
[url]https://github.com/EdgeTX/edgetx/blob/5d9498e88be70cb8d34bac4f08bcac8cc9e5452f/radio/src/targets/common/arm/stm32/adc_driver.cpp#L145[/url]

but never disabled, causing unnecessary strain on the battery.

If on the other hand _Check RTC voltage_ is enabled, after first checkRTCBattery() call:
[url]https://github.com/EdgeTX/edgetx/blob/5d9498e88be70cb8d34bac4f08bcac8cc9e5452f/radio/src/opentx.cpp#L695-L703[/url]
disableVBatBridge() is called:
[url]https://github.com/EdgeTX/edgetx/blob/5d9498e88be70cb8d34bac4f08bcac8cc9e5452f/radio/src/targets/common/arm/stm32/board_common.h#L120-L123[/url]
removing the bridge.

Tested on RM TX16S - one sees jumpy values of RTC Batt when starting a radio with _Check RTC voltage_ disabled - meaning the value is sampled constantly. On the other hand when enabling _Check RTC voltage_, reboot and then checking the RTC Battery value - it is rock solid, thus being sampled only once as it should be.

rotorman, 2021-06-25:
See also https://github.com/opentx/opentx/issues/8571

rotorman, 2021-06-27:
I created a current measurement rig for coin cell on RM TX16S by soldering a second CR 1220 holder in parallel, where I put the battery and then putting a current meter in series to the pinheader on the positive rail:

[image]

My current meter can do in most sensitive range 400µA measurements with 100nA resolution. Testing with 5d9498e8.
When radio is off, I measure 0.7 to 0.8 µA, likely the current is even lower, just my device cannot do better.

When radio is powered on and under SYSTEM -> HARDWARE -> "Check RTC voltage" is ticked, the voltage reading behind _RTC Batt_ stays constant and the measurement reads still ca. 0 to 0.1 µA (thus less when radio is off).

When on the other hand "Check RTC voltage" is NOT checked, I see _RTC Batt_ value toggle and read 24.9 to 25.0 µA on the current meter.

pfeerick, 2021-06-27:
And let me guess... it's not a rechargeable coin cell? So in "checking" the RTC battery voltage, it's actively being flattened? 😢

raphaelcoeffic, 2021-06-27:
> And let me guess... it's not a rechargeable coin cell? So in _not_ "checking" the RTC battery voltage, it's actively being flattened? 😢

Exactly!

rotorman, 2021-06-27:
Solved by #360.

PR #360 (2021-06-28): RTC bridge fix
Code thanks to @bsongis https://github.com/opentx/opentx/pull/8577 and original tip by @MikeBland https://www.rcgroups.com/forums/showpost.php?p=47360301&postcount=82 and https://github.com/opentx/opentx/issues/8571

Resolves #333. After applying this fix, I re-measured the current draw on the RM TX16S coin cell.
Irrespective of the setting under SYSTEM -> HARDWARE -> _Check RTC voltage_ now, the reading behind RTC Batt stays constant and the measurement reads ca. 0 to 0.1 µA.

Briefly on power-up I see 25µA, as the voltage is being sampled once (irrespective if the _Check RTC v [... cut]
closed 2021-06-27 by rotorman