EdgeTX/edgetx #334: eachine t16s

pds59, 2021-06-25:
The issue tracker here is for devlopment of EdgeTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/), [Discord](https://discord.gg/wF9wUKnZ6H)
or [Facebook](https://www.facebook.com/groups/edgetx).

Issues that fall into the category of asking for help may be re-directed
to the other support platforms or answered directly if time permits.

pds59, 2021-06-25:
after upgrading to edge tx,  volume is about 70% of original open tx version no matter if it is set to max

rotorman, 2021-06-25:
This must stem from the sound files on the SD card, as the Digital-Analog-Converter (DAC) driver code for OpenTX and EdgeTX is exactly the same (not a single char difference):
https://github.com/opentx/opentx/blob/2.3/radio/src/targets/common/arm/stm32/audio_dac_driver.cpp
vs
https://github.com/EdgeTX/edgetx/blob/2.4/radio/src/targets/common/arm/stm32/audio_dac_driver.cpp

pfeerick, 2021-06-26:
In other words - is this comparison with the same sound pack? As if you are comparing different sound packs, it's not a valid comparison/issue for this repo as this isn't firmware related. 

If you are making a comparison between the OpenTX sound pack and the EdgeTX one, please ensure you are using the latest version of the sound pack also, as there was a rc3 release two days ago now which normalised the audio levels, which should make them consistent in volume, and also somewhat louder.  I you are running the latest version of the voice pack, and it's still noticably quiet, raise an issue [on the issue tracker for the voice pack](https://github.com/EdgeTX/edgetx-sdcard-sounds/issues) for the voice pack.

pds59, 2021-06-26:
Solved was the same sound file but something was wrong i uploaded a new One and now Is fine tksPaolo de scaglia

closed 2021-06-26 by pfeerick