EdgeTX/edgetx #335: Executing a widget from SD card flashes screen briefly with noise

rotorman, 2021-06-25:
When executing any LUA widget (main.lua) under SYSTEM -> SD-HC Card -> WIDGETS subfolder, the screen briefly flashes and displays noise:

Here a short video demonstrating this (running 5d9498e8 on RM TX16S):
https://www.youtube.com/watch?v=sVrfNoy1oZQ

pfeerick, 2021-06-26:
Is it just the those widgets? I notice it also did't ... for a better word... run. Is it something to do with them not being 'standalone' scripts? Maybe a warning should appear saying they're not 'runnable' directly? I noticed something somewhat similar when I was working an a simple lua app, and it seemed to have a connection to how soon I cleared the screen (it was a fullscreen app).

rotorman, 2021-06-26:
> Maybe a warning should appear saying they're not 'runnable' directly?

Yep, that would already be a good solution, together with a suppression of the display flash.

philmoz, 2023-12-13:
Is this still an issue?

The 'noise' has been replaced by the 'loading' message.
Trying to run a widget directly will show an error, that the 'run' function is missing.

pfeerick, 2023-12-13:
I don't think so. The noise was part of the issue, and 2.6 onwards shows an error if you try to run a widget manually (issue was raised during the 2.4 days) so yup, bye bye! :)

closed 2023-12-13 by pfeerick