EdgeTX/edgetx #338: Timer 1 and 2 activate with  Thrtl channel override active.

MervDale, 2021-06-26:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 
I use Timer 1 as a total flight log ( no reset)
I use Timer 2 as current flight time (battery time, approx )  with reset

Both timers still acivate with throtle movement with channel override switch active
Start mode set to Throttle  switch SC UP  set to overide Channel 2    -100  Motor OFF
SC MID  SC DOWN = Motor ON 

I have checked this with OpenTx 2.3.12 and 2.3.13 and they have the same issue with the timer activated with the THs  option, Not sure  if this occured on earlier version of OpenTx 

I would have thought that the option of starting a timer with the Throttle activation, 0 upwards should be deactivated if the throttle cannel is over ridden ?

[image]

pfeerick, 2021-06-26:
I'm wondering if this a logic problem more than a bug... 

If I understand this right, you have SC it so that in the up position is overriding the throttle channel to -100 (i.e. throttle stick down). The simplest way to do this would be to have the !SCup (_not_ SC up) position trigger the throttle override. 

It would then make sense that if you want the timers to trigger on throttle start, that they want to be set to the !SCup switch state, so that they are also can only start counting when SC is _not_ in the up position. 

Throttle start will start counting, until you reset it, so when you arm, and move the throttle, the timer will start and stop when arm/disarm via SC. If you use Throttle as the mode, the counter will only run when armed, and the throttle stick is moved above -100.

Please don't hesitate to correct me if I've wrongly assumed something here... But for me, this starts the timers if the throttle stick is moved AND SC is in the mid or down position, stops them when the switch is in the up position, but they start again on their own if the switch is then moved back into the middle or down position. That last part could be prevented by using Throttle instead of Throttle Start. 

[image]
[image]

[image]

MervDale, 2021-06-26:
Ok, this works but the Throttle switch need to be set to "Throttle"  not "Throttle Start".

pfeerick, 2021-06-27:
Excellent. Yes, with timer mode set to Throttle Start the timer will start the first time the throttle moves (and the switch condition is true), so if you disarm/re-arm by moving SC up and then down again, the timer would resuming counting regardless of the throttle position. This is exactly how THs behaves on OTX... so is not a new or different behaviour.

closed 2021-06-27 by pfeerick