EdgeTX/edgetx #342: Insta trim not working

Mantamatt1, 2021-06-26:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-06-26:
With what Transmitter? What firmware version? How exactly are you setting up InstaTrim? Special Function for the model, Global function? The questions in the template are so we don't need to guess those things! ;)

I can't reproduce this on RC3 with either the X9D+ or T-Lite, with Inst.Trim attached to a switch, as a model special function. 

I *can* reproduce it on the TX16S, again with Inst.Trim attached to SFdn, as a model special function (setting it up as a global function does not work either). The functions screen indicates the function is active (i.e. row goes yellow when switch in trigger position, but the channel output is not adjusted "trimmed" to match).

Mantamatt1, 2021-06-26:
Yes tx16s, sf down, function goes yellow but trims do nothing

Mantamatt1, 2021-06-26:
I have re commented.
It's a tx16s, sf down. Function goes yellow but no movement on trims.

Thanks for all your hard work, you have done amazing things

Mantamatt1, 2021-06-26:
Firmware is the latest rc3

raphaelcoeffic, 2021-06-26:
As it seems, Insta trim is meant to be enabled only when certain menus / screens are displayed:
```
#define IS_INSTANT_TRIM_ALLOWED()      (IS_MAIN_VIEW_DISPLAYED() || IS_TELEMETRY_VIEW_DISPLAYED() || IS_OTHER_VIEW_DISPLAYED())
```

Now, with the new UI, it is not known globally which view/screen is displayed, so it seems it has been disabled. Does anyone know why? Or what is the idea behind that?

eshifri, 2021-06-26:
In functions.cpp:

          case FUNC_INSTANT_TRIM:
            newActiveFunctions |= (1u << FUNCTION_INSTANT_TRIM);
            if (!isFunctionActive(FUNCTION_INSTANT_TRIM)) {
#if defined(COLORLCD)
  #warning IS_INSTANT_TRIM_ALLOWED() is always false
#endif
              if (IS_INSTANT_TRIM_ALLOWED()) {
                instantTrim();
              }
            }
            break;

raphaelcoeffic, 2021-06-26:
> In functions.cpp:
>
> ```
>       case FUNC_INSTANT_TRIM:
>         newActiveFunctions |= (1u << FUNCTION_INSTANT_TRIM);
>         if (!isFunctionActive(FUNCTION_INSTANT_TRIM)) {
> ```
>
> #if defined(COLORLCD) [...]

I'm just wondering why on b/w radios the function is limited to certain views. This is quite strange to me.

eshifri, 2021-06-26:
I guess that if you are in the model setup views, you do not want to invoke this function by mistake.

pfeerick, 2021-06-27:
What counts as an `IS_OTHER_VIEW_DISPLAYED` though... on X9D+ with ETX (and on OTX 2.3.12 using the sim) I can trigger Insta Trim as a global function in both radio setup and model setup... I'm not immediately aware of where it doesn't work... 

I personally think it should be enabled, enabled until it's disabled... it shouldn't be left active all the time, so having it work conditionally just adds confusion and potential for it to be mistakenly left on.

raphaelcoeffic, 2021-06-27:
> I guess that if you are in the model setup views, you do not want to invoke this function by mistake.

Doesn't that fall into "be a responsible user"? If you do things by mistake, you need to cope with the consequences ;-) I could however understand having a sound telling you you just used "Insta. trim". Noticing which menu you are currently in with in the new UI is a non-trivial task, and would require quite some effort. So I'd advocate for letting down that "security feature" and just let it be a special function you can invoke, whatever is displayed on the screen.

Mantamatt1, 2021-06-27:
I use insta trim all the time with otx.
I activate it by ticking the box.
I maiden the model.
When using instatrim it beeps every time you press it (that's quite
important). I land and un tick the box to deactivate it just incase I
accidentally press it it!

Its totally my responsibility if I leave it activated and accidentally
press it.

You cant fix stupid! 😁

pfeerick, 2021-06-27:
> You cant fix stupid! 😁

I think we're all in agreement... if you insta crash it you insta to blame ;)

raphaelcoeffic, 2021-06-27:
> > You cant fix stupid! 😁
> 
> I think we're all in agreement... if you insta crash it you insta to blame ;)

Should we post this on « Insta. Gramm »?

PR #347 (2021-06-27): [color] enable insta trim on all views
Resolves #342
commit adde1160 (2021-06-27): [color] enable insta trim on all views (#347)

* [color] enable insta trim on all views

Resolves #342

* Removed warning for colorlcd
closed 2021-06-27 by raphaelcoeffic