EdgeTX/edgetx #343: Spectrum ar610 receiver problem

Mantamatt1, 2021-06-26:
On rc2 Spektrum ar610 would only work on dsm2 1f and no out put to Channel 5 
Other spektrum receivers were not affected. 
Now rc3 ar610 works on all channels but still only on dsm2 1f

Mantamatt1, 2021-06-26:
Tx16s transmitter

raphaelcoeffic, 2021-06-26:
@Mantamatt1 Could you please try to ask the Multi-protocol people what the issue could be? I'm a bit puzzled right now, and do not own such hardware to try. I'm not saying EdgeTX is not involved, but it will be hard for us to debug, so that we might have a better chance if Pascal looks at it.

Mantamatt1, 2021-06-26:
Yes I will ask them. I popped this bug to yourselves as changing to edgetx
is the only change I done, then updating the module done no different

pfeerick, 2021-06-27:
It's more a puzzling issue, because there hasn't been any *specific* changes that would make this happen... even more perplexing is that it seems to be specific to a particular model of receiver... if it affected DSM2 generally, I'd be more certain it's something in ETX, but that seems just plain weird.

raphaelcoeffic, 2021-06-27:
> Yes I will ask them. I popped this bug to yourselves as changing to edgetx is the only change I done, then updating the module done no different
> […](#)

Could you please re-try with the updated module (same firmware as right now) and OTX? I have the feeling this could be related to the module firmware itself.

Mantamatt1, 2021-07-22:
Latest mpm firmware and otx works.
 I had forwarded this issue to the multi protocol people but have had no response!  What's the next step?

richardclli, 2021-07-22:
> 
> 
> Latest mpm firmware and otx works.
> I had forwarded this issue to the multi protocol people but have had no response! What's the next step?

Please specify the versions that you have tried works and the versions that are not working, including mpm and radio.  I am familar with mpm as well, maybe I can take a look in it.

Mantamatt1, 2021-07-22:
Tx16s
Etx on latest as of yesterday rc4
Receiver spektrum ar610 dsmx
Mpm v1.3.2.86 taer
Receiver binds and works fine on dsm 2 1f
But gets all messed up binding to dsm 2 2f
Binding with dsm x 1f all channels delayed movement.
Binding with dsm x 2f is same as dsm x 1f
All binds have no servo movement on ch5.
Channel monitor is showing output

Mantamatt1, 2021-07-22:
This has been the same since rc2 and the previous mpm version

raphaelcoeffic, 2021-07-24:
> Tx16s
> Etx on latest as of yesterday rc4
> Receiver spektrum ar610 dsmx
> Mpm v1.3.2.86 taer
> Receiver binds and works fine on dsm 2 1f
> But gets all messed up binding to dsm 2 2f
> Binding with dsm x 1f all [...]

Thx for reporting. I happen to have a AR610 laying around, so I can check how it behaves.

pfeerick, 2021-07-24:
Maybe also look at the number of channels involved - as [something seems to change above and below 7/8 channels](https://www.rcgroups.com/forums/showthread.php?2579769-6-Channel-Limit-for-1024-22ms-Mode) - i.e.  is the channel range restricted to CH1 - CH7?

gagarinlg, 2023-03-18:
@frankiearzu do you know if this is still a thing?

frankiearzu, 2023-03-18:
The sub-protocols tells the servo pulse length (1F=11ms  2F=22ms) as well as Telemetry (on/off) 
From what i can tell in the EdgeTX code, we really don't do much, except passing the data to the multi-module. Only in the case of sub-protocol AUTO, we update the returned sub-protocol.
Will check the code in the multi-module to see if there is something there.
I don't think that i have a AR610, but i think i have an old one supporting only DSM2, will try,

gagarinlg, 2023-09-18:
Does anyone have an update on that issue?

gagarinlg, 2023-09-20:
closing this for now, as by @frankiearzu s recommendation

closed 2023-09-20 by gagarinlg