EdgeTX/edgetx #344: Voice seconds countdown mixed up

Mantamatt1, 2021-06-26:
Tx16s 
Firmware rc3
Timer count down voice seconds 5s 10s 20s 30s is mixed up eg... 30s= 5s countdown. 5s = 30s countdown etc

jogybaer0815, 2021-06-26:
TX16S RC3 - not for me - i have installed sound pack rc3

Mantamatt1, 2021-06-26:
I am using my original sounds from opentx. Would this be causing the
problem? As I was aware it's just a basic wav file. Suppose the naming of
the system files could be mixed? Simple fix if that's all it is as I'll
just rename the files to suit

pfeerick, 2021-06-27:
Seems like the voice files must be wrong somehow... I got the correct 30, 20, 5, 4,3,2,1, timer one elapsed sounds... using the ETX voice pack. However, checking the files on both ETX and OTX voice packs suggests there should be no difference, and I don't know of any a changes to the SD card pack that would explain that difference. 

2.4.0-RC3, TX16S

MervDale, 2021-06-27:
Countdown timer call settings are back to front

Eachine TX16S  Timer 2 settings

set 5 = time counter call starts at 30

set 10 = time counter call  starts at 20

set 20 = time counter call starts at 10

set 30 = time counter  call starts at  5

pfeerick, 2021-06-27:
The above seems to be a different issue (so is in its own issue), but I can repeat that on the TX16S.

Mantamatt1, 2021-06-27:
Yes it's the same issue, maybe I worded it badly!

Mantamatt1, 2021-06-27:
Mine was doing it on timer 1. Never tried timer 2

pfeerick, 2021-06-27:
Ah, sorry I thought you mean it was saying the wrong numbers! 🤦‍♂️ 

In that case, I can repeat it, and it's on both Timer 1 and Timer 2.

rotorman, 2021-06-27:
Plz test #362 - that should fix it.

Mantamatt1, 2021-06-27:
Just tested and is working as it should

Mantamatt1, 2021-06-27:
On tx16s

rotorman, 2021-06-27:
Thx for testing, closing.

pfeerick, 2021-06-27:
It may be a style thing, but please leave issues open until the PRs are merged. As long as you use the keywords Fixes, Closes, Resolves #issue-id in your commit msg or PR, github will auto-close the issue WHEN the issues is actually fixed ;) If something goes wrong and the PR isn't merged, well, this isn't fixed and still open, isn't it ;)

PR #362 (2021-06-28): Voice timer countdown fix
I had a look with hw-debugger at the values saved to **signed 2 bit integer** _timer[x].countdownStart_ at various UI dialog set values
https://github.com/EdgeTX/edgetx/blob/831ec1751ccfb381f4e36d762dc6ed3210d39bed/radio/src/gui/colorlcd/model_setup.cpp#L1061

- 5 sec saves value -2
- 10 sec value -1
- 20 sec value 0
- 30 sec value 1

The values are evaluated originally by wrong conversion to seconds by:
https://github.com/EdgeTX/edgetx/blob/831ec1751ccfb381f4e36d762dc6ed3210d39bed/radio/src/myeeprom.h#L224-L228
that is used by audio output:
https://github.com/EdgeTX/ [... cut]
closed 2021-06-28 by raphaelcoeffic