EdgeTX/edgetx #345: Last used model

Mantamatt1, 2021-06-26:
Tx16s 
Firmware rc3
When selecting a  model to go to last used model not to the beginning of the list

eshifri, 2021-06-26:
It is the same in openTX, isn't it?

jogybaer0815, 2021-06-26:
No if i select a model in OpenTX it leaves the list (it sends a return). This should be better than remaining in the list

eshifri, 2021-06-26:
I'm sure it is like this on Taranis: selected model is just marked by "v"
It is the same in NV14 version, but it is something not standard at all.
I do not remember how it worked on TX16S in OTX2.3.11, but your description definitely makes sense.

eshifri, 2021-06-26:
It also means you cannot duplicate current model, so I'm not sure...

Mantamatt1, 2021-06-26:
Right ok. Just annoying as I've 40+ models and the latest in the list are
my most used models, and no way to arrange the list so with the touch
screen I have to scroll to the bottom of my list  to change to another
frequent model. As the scroll wheel doesnt have a quick scroll option it's
a bit of a pain.
Talking about quick scroll while scrolling through sound files its it is
very slow as it doesnt to that skip thing when scrolling fast!

Images... can a preview be added to the list so i can see the selection
rather than just the text!

Thanks guys

eshifri, 2021-06-26:
Which preview are we talking about?
SD card manager shows image in preview.

So you want an option to move model tp the top of the list?

Mantamatt1, 2021-06-26:
Sd card shows a preview but when selecting an image in the widgets I would
like to see a preview.

Models list... to be able to arrange, for example options maybe..., ability
to reverse order, alphabetical, manual, list option, icon size smaller,
fast scroll etc

pfeerick, 2021-06-27:
> 
> 
> It is the same in openTX, isn't it?

No, and it doesn't prevent you from duplicating the model either. 

In OTX, when you select (as in load, not highlight) the model, it jumps out of the menu, to the model... in ETX, it stays in the menu, which would sort of be ok, if it also keped the just activated model in view... which it doesn't. 

i.e. this is the OTX behaviour... 
[image]

Also, can we stick to one issue per issue please... ;) SD the sdcard browsing card stuff is not related to the original issue ;)

pfeerick, 2021-06-27:
Also, isn't you request at #348 really what should happening here?

Mantamatt1, 2021-06-27:
#348 was a different issue. I was referring when entering the models page it starts at the beginning of the models and have to scroll to the bottom every time I fly a different model as my newest models and most likely models are usually the latest ones in the list. Example... today I was flying model 43 and 45 but every time I went to fly the other model its starts on model 1 and I have to swipe/scroll through 40+ models to get to the one I want to fly. 
#345 refers to when I select the model I want to fly ie number43 the page reverts back to model 1, expected behaviour would be to go to the home screen for the model i just selected

pfeerick, 2021-06-27:
Ah yes, I get what you mean now. Sorry about that! I agree, and you're not the first person to ask for that. I can't see where it was now, I thought it was another issue but it was probably a request on Discord, and it is certainly something that should be fixed. Hopefully it's 'just' a matter of making the view automatically scroll down to the active model. ;)

raphaelcoeffic, 2021-06-27:
> Ah yes, I get what you mean now. Sorry about that! I agree, and you're not the first person to ask for that. I can't see where it was now, I thought it was another issue but it was probably a [...]

Exactly this. I added the code in #355.

raphaelcoeffic, 2021-06-27:
@Mantamatt1 Please try the firmware in #355 and tell me if this is what you mean ;-)

Mantamatt1, 2021-06-27:
Yes! That's perfect! Spot on!
So is this option going in the build or is it just specific to my request?

raphaelcoeffic, 2021-06-27:
> Yes! That's perfect! Spot on! So is this option going in the build or is it just specific to my request?
> […](#)

If this is what you meant, we should merge it after some tests, and then it’s in main version.

Mantamatt1, 2021-06-27:
Sounds good
Thanks for all your swift hard work

PR #355 (2021-06-28): [color] set focus on current model or close to the current operation
Resolves #345
commit 79eee634 (2021-06-28): [color] set focus on current model or close to the current operation

Resolves #345
closed 2021-06-28 by raphaelcoeffic