EdgeTX/edgetx #349: more overview in MODEL SELECT due to smaller icons

jogybaer0815, 2021-06-26:
TX16S RC3
If I have more than 6 models in the respective directory, it quickly becomes difficult to scroll through the large icons to the respective model. 
Would it be possible to freely select the icon size and / or the number of rows / columns?
Thanks

pfeerick, 2021-06-27:
That's a question for @raphaelcoeffic  .... I would suggest not in the first instance, as that would require resizing the model images every time they are displayed, and also hint that there is some discussion as to changes to that UI, which might satisfy you if that comes to fruition. But you didn't hear it from me. 😉

raphaelcoeffic, 2021-06-27:
@pfeerick this is in fact a question to @JimB40, who designed that screen ;-)

JimB40, 2021-06-27:
Short answer is 'Don't go this way'. :) 
Surely not in upcoming iterations. Scaling will introduce lot of UI issues. Like text clipping etc.
1. Model selection screen will be re-worked in 2.4.1
2. Labels and filtering are to be introduced to ease model selection
To be closed.

pfeerick, 2021-06-27:
Our UI/UX expert guru has spoken... stay tuned, but for now we won't be going this way just yet ;)

closed 2021-06-27 by pfeerick