EdgeTX/edgetx #351: Global widgets

Mantamatt1, 2021-06-26:
Tx16s
Every time I create a new model I have my basic setup for all models I.e model name, timer outputs ect.
Can this be setup as a global thing so when i create a new model it's not just a blank page. 

I know you can make a template and copy it but I prefer to create a new model as I dont have to hunt through the list for the template later down the line

lshems, 2021-06-27:
I would really suggest to make a folder with your preferred 'templates'.

You now ask for 'global' widget setup. As soon as this exists, you want to be able to 'switch' between global setups.

Something which is perhaps also covering your needs, is to be able to set one of your models as a default to copy for new models, with a name 'new' & oldmodelname.

This would be WAY easier to implement and cover the same needs.

Just my counterproposal to cover in a simpler way your request.

Mantamatt1, 2021-06-27:
Ok but for now there is no folder option.

raphaelcoeffic, 2021-06-27:
Templates would be a good idea indeed. The only that bothers me is that the values related to telemetry are indexed by sensor ID rather than name, which is an issue (the „Value“ widget).

lshems, 2021-06-27:
Yes, there is? Or not in edge? In opentx there is?

lshems, 2021-06-27:
And the sensors, if you allow me, there is a solution, but it comes with drawbacks. Drawbacks I don't think are really blocking. I'll make another thing for that.