EdgeTX/edgetx #353: Touchscreen - Randomly non responsive - Reboot radio to restore

mwilson007, 2021-06-27:
I have a Radiomaster TX16s running EdgeTX 2.4 rc3.

I am getting random touch screen lockups and when this happens all other radio functions are still fine and I can use the buttons and jog wheel to navigate.

The only way to get touchscreen functionality back is to reboot the radio.

pfeerick, 2021-06-27:
This is unfortunately a known issue, and we're still trying to pin it down whether it is a hardware issue in the touchscreen controller, or a software issue. The most annoying part is that it won't happen when you try to reproduce it, and it will happen when you finally think it's gone! 🤪 We *think* it may be a software issue, as recent more repeatable touchscreen locks appear to be happening when there are at least two widgets in the top layout bar? Is this the case for you?

If it happens again, and you still have your screen set to dim after a delay (Radio setup -> Backlight, Off brightness is lower than On, mode is Both, delay is 10 sec)... please leave the radio alone for longer than that delay and see if it dims. If it does, does it brighten if you touch the display (don't press the buttons, move the stick... we're checking if the hardware interrupt is triggering)... if it doesn't brighten, it's mostly likely a problem with the touchscreen chip, if it doesn't dim at all, or pressing it 'does' brighten the screen, it's a software issue. And naturally, report your results! :)

mwilson007, 2021-06-27:
I have 4 widgets in my top bar.  I have seen random touchscreen lockups when setting up widgets as well as just when using random radio functions.

My backlight settings are as you stated above and it will dim after 10 seconds and then brighten if I touch the screen.

pfeerick, 2021-06-27:
Nice... that sounds like it's software, and we're getting closer. If you get a chance, can you also play with a model that is basically the same (maybe just duplicate your current one to minimise variation, and remove the top bar widgets... ) without top bar widgets, and see if the touch screen locks up less?

mwilson007, 2021-06-27:
Yes, I have some models that don't have any top bar widgets.  I will play around with those to see if the lockups are less frequent.  Note that I received lockups when setting up the radio initially after upgrade before I had any top bar widgets on any of my models so I don't think it's primarily related to the widgets.  I have tried to see if I can get lockups at will but I can's seem to easily replicate the issue.

pfeerick, 2021-06-27:
Yeah, it's infuriating... I got it to lock up twice in a row (IIRC, it was a clean slate since I'd let the initial version of the flasher erase the whole SD card so had nothing setup at all) when going into the radio setup... and then it just wouldn't / hasn't done it again... 🤷‍♂️

mwilson007, 2021-06-27:
Yeah, I had 4 random lock ups during my model setups after the upgrade but I have not been able to repeat it purposely.

ZZ-Cat, 2021-06-28:
I was having a bit of a play around with my TX16S (making some adjustments on all of my model entries), & the only time I've had the touch screen freeze on me like that, is whenever a.) I have a model that has widgets in the top bar of the main page; & b.) Whenever I've gone into any of the widgets settings & made some changes there.

Whenever I'm changing the widgets' settings, that, to me, is where I can _somewhat_ reproduce that lockup consistently.

I do have my hopes up that it's a software issue with you guys, & _not_ an issue with the touch screen's driver itself.

raphaelcoeffic, 2021-06-28:
> I was having a bit of a play around with my TX16S (making some adjustments on all of my model entries), & the only time I've had the touch screen freeze on me like that, is whenever a.) I have a [...]

The "backlight test" is decisive here. As long as the backlight comes back when you touch, the sensor is still responsive.

mwilson007, 2021-06-28:
I believe my issues were also encountered when editing widgets and scrolling through model settings.  I have not been able to recreate at will and it has not happened since my initial setup.

raphaelcoeffic, 2021-07-02:
@mwilson007 @ZZ-Cat could you please try the code in #378? I believe there is a good chance of fixing at least a good part of the issue.

ZZ-Cat, 2021-07-02:
> @mwilson007 @ZZ-Cat could you please try the code in #378? I believe there is a good chance of fixing at least a good part of the issue.

Right-oh.
Did this get merged with one of the nightly builds? If not, how do I go about testing it?

raphaelcoeffic, 2021-07-02:
> > @mwilson007 @ZZ-Cat could you please try the code in #378? I believe there is a good chance of fixing at least a good part of the issue.
> 
> Right-oh.
> Did this get merged with one of the nightly builds? If not, how do I go about testing it?

No, it's not yet merged. You can grab the firmware compiled on the "Checks" tab, then look for "Artifacts". Please make sure you're logged into GH otherwise you won't see the artifacts.

ZZ-Cat, 2021-07-02:
> No, it's not yet merged. You can grab the firmware compiled on the "Checks" tab, then look for "Artifacts". Please make sure you're logged into GH otherwise you won't see the artifacts.

Sweet as. 👍
I've got it now, & I'll let ya know what's up, in a few minutes.

ZZ-Cat, 2021-07-02:
@raphaelcoeffic 
Well, I got it done on my TX16S Max Edition, & of all of the possible ways that I could think of to get the touch screen to lock up... I couldn't get it to lock up.

I spent a good 2 hours on my RadioMaster, trying to get the screen to lock up; & it didn't.
I've come to the conclusion that if the touch screen was gonna lock up, it would have done so, by now.

phileville, 2021-07-02:
I got a report from a user on FB that they still had a lockup testing with the  touch polling fix:
"Karl Heinz

Toch-polling doesn`t fix this problem, after approx. one hour working in model-settings, the screen locks again!

raphaelcoeffic, 2021-07-02:
> I got a report from a user on FB that they still had a lockup testing with the touch polling fix:
> "Karl Heinz
> 
> Toch-polling doesn`t fix this problem, after approx. one hour working in model-settings, the screen locks again!

Any feedback on the „backlight test“? This is very important, as this could be related to another issue where the sensor really stopped working.

brainbubblersbest, 2021-07-02:
This can be pure randomness. But the screen on my x12s today freezes too. Need to shut it off and reboot. Maybe this is not really related to Touch???

raphaelcoeffic, 2021-07-02:
> This can be pure randomness. But the screen on my x12s today freezes too. Need to shut it off and reboot. Maybe this is not really related to Touch???

I believe this is another issue. Under certain circumstances, the focus might get placed on components that do not process keys, so that it feels like it is frozen. However, as long as you can use the power button, it is not really frozen.

brainbubblersbest, 2021-07-02:
The screen itself showed no sign of reaction. Page Buttons rtn MdL and sys seems to be dead. Never had something similar on that Radio and this showed up pretty quick after first Minutes using edge. Just some Feedback 😉

raphaelcoeffic, 2021-07-02:
> The screen itself showed no sign of reaction. Page Buttons rtn MdL and sys seems to be dead. Never had something similar on that Radio and this showed up pretty quick after first Minutes using edge. Just some Feedback 😉

No problem! I‘m just trying to classify issues properly so we don‘t mix them up.

mwilson007, 2021-07-02:
just doing some setup changes to the model and had a touch screen lockout again.  This time the backlight was unresponsive when I touch the screen as well.  I'm not running the polling fix firmware.

raphaelcoeffic, 2021-07-03:
> just doing some setup changes to the model and had a touch screen lockout again. This time the backlight was unresponsive when I touch the screen as well. I'm not running the polling fix firmware.

What do you mean by unresponsive? Was it ON all the time, or was it off after a while, and never went back by touch input?

mwilson007, 2021-07-03:
The backlight stayed bright all the time and would not dim again until I restarted the radio. After that, the backlight and touchscreen worked fine again.

raphaelcoeffic, 2021-07-03:
> The backlight stayed bright all the time and would not dim again until I restarted the radio. After that, the backlight and touchscreen worked fine again.

Thx. This issue should be fixed by the touch fix PR. This is the one I could somehow reproduce the easiest, and did not come back.

mwilson007, 2021-07-03:
ok cool, thx

closed 2021-07-03 by raphaelcoeffic

mwilson007, 2021-07-03:
to get the polling fix, should I just flash the latest nightly build?

raphaelcoeffic, 2021-07-03:
> to get the polling fix, should I just flash the latest nightly build?

That will work tomorrow. I just merged the PR. In a couple minutes, it will be available as well from the flasher if you take the latest build on 2.4.

mwilson007, 2021-07-03:
ok thanks.  I just used 6c1e418 in the meantime.