EdgeTX/edgetx #356: range mode RSSI

weixelgeist, 2021-06-27:
EdgeTX 2.4.0 RC3
Jumper T18

in range mode the RSSI is not shown

raphaelcoeffic, 2021-06-27:
> EdgeTX 2.4.0 RC3
> Jumper T18
> 
> in range mode the RSSI is not shown

I guess this is with the internal Multi module, right?

weixelgeist, 2021-06-27:
> > EdgeTX 2.4.0 RC3
> > Jumper T18
> > in range mode the RSSI is not shown
> 
> I guess this is with the internal Multi module, right?

Yes

pfeerick, 2021-06-27:
As in, the RSSI screen is not shown? As OTX used to bring up a screen that just had RSSI on it... ETX seems to trigger the range mode, but doesn't bring that screen up. 

[image]

eshifri, 2021-06-27:
What if we just output RSSI on the Range button (instead of the word "Range"), while it is yellow and blinking?
Or is it too small and you prefer the whole Page?

pfeerick, 2021-06-27:
I think if you were outside in the sunlight it would be too small, as it's a pretty small button. I don't *mind* the whole page, but also feel it's too big. However, I think part of the purpose is so that you have to use the RTN button to exit it, thus in doing so stopping the range check. Otherwise, at the moment, you could leave the screen, return to it, and while the UI doesn't indicate the range check is still active, pressing the button again stops the range check.

pfeerick, 2021-06-28:
What 'could' be done is show it all the time, which actually brings up another bug - when doing frequency tuning for certain protocols (i.e. Frsky), the RSSI is supposed to be shown, but it isn't. 

[image]

eshifri, 2021-06-28:
But it is still the same small size, isn't it?

pfeerick, 2021-06-29:
Yeah, that's why it's not ideal if it was the only way to see RSSI whilst in range test mode. That RSSI display is for a different reason - calibration of the internal multiprotocol module - and you generally don't touch it again after doing the initial calibration (which will probably be indoors) so isn't such a problem. The range test is likely to be outdoors if you're following procedure and range-checking before flight (which we all do, right? 😁) ... so it would be better to not make it hard to see ;)

closed 2021-07-01 by raphaelcoeffic