EdgeTX/edgetx #359: filmware

pds59, 2021-06-27:
The issue tracker here is for devlopment of EdgeTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/), [Discord](https://discord.gg/wF9wUKnZ6H)
or [Facebook](https://www.facebook.com/groups/edgetx).

Issues that fall into the category of asking for help may be re-directed
to the other support platforms or answered directly if time permits.

pds59, 2021-06-27:
sorry for primitive question, how to understand in the flasher the latest firmare branch and version ?
tks in advance

closed 2021-06-27 by pds59

pfeerick, 2021-06-27:
I hope that means you found the answer. 

The flasher is in early stages of development, but there is a RC3 entry in the firmware branches menu in the flasher that should at least point you the the right firmware.