EdgeTX/edgetx #364: External uarts for receiver flashing

Mantamatt1, 2021-06-27:
Just an idea! On my old qx7 I used the external serial port for flashing receivers but on the tx16s I have to remove the external module to flash off the pins. Is it possible to implement the external ports on the botton of the controller for flashing receivers as I think it's a much cleaner convenient way

3DRacer, 2021-06-28:
Another advantage would be the stabilized 5V at this connection (with the TX16S). This means that a Zener diode would no longer be necessary to prevent 5V receivers from grilling on the module bay.

pfeerick, 2021-06-28:
I think I saw mention in one video that it wasn't possible for a technical reason, maybe it was that - the inability to reset - as I would have though it was just connected directly to the power rails... but I don't know. @rotorman would most likely know if it's actually technically feasible to flash a receiver via those AUX1 and AUX2 ports...

rotorman, 2021-06-28:
I am not aware of the protocol or details used to flash receivers, but the RM TX16S AUX1 and AUX2 are 3.3V logic level no-hw-flow std. non-inverted UARTs/serial ports and the +5V rail can be switched on/off from firmware. Thus if you just need bi-dir serial port and power cycling via +5V rail would be enough, it should IMHO work.

[image]

The 5V rail is controlled via Q120 for AUX1 and Q121 for AUX2. The STM32F429BIT6 pins for power control are pin 160 PA15 for AUX1 and pin 61 PB0 for AUX2.

More info here: https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram (Sec. 2.8)

raphaelcoeffic, 2021-06-28:
> I am not aware of the protocol or details used to flash receivers, but the RM TX16S AUX1 and AUX2 are 3.3V logic level no-hw-flow std. non-inverted UARTs/serial ports and the +5V rail can be [...]

I think the point here is that FrSky receivers need a one wire half-duplex inverted connection, basically meaning that this needs to implemented in software. Not a big deal, but someone needs to do it.