EdgeTX/edgetx #366: New models to fill gaps of deleted models

Mantamatt1, 2021-06-28:
Tx16s 
Rc3 but unchanged from otx
Not sure if this is a feature request or a bug?

When you create a new model it has a model number I.e model25 ( I use this number as the receiver number aswell. (wish it would do it automatically).
When you delete a model let's say model15
And create a new model it will create model26

It doesnt fill in the gaps so model15 would just be a gap and the model numbers keep going up and up. 
On my qx7 I could fill in the gaps and receiver number would match and not be random manual selection

pfeerick, 2021-06-28:
I would say if it's a behaviour from OTX, then we can call both a fix of their bug, as well as an 'enhancement' ;) I like the idea - if only from a OCD / neat freak standpoint... since there's going to be gaps in the filenames, as well as the 'blank' titles ;)

Mantamatt1, 2021-06-28:
Its definitely an ocd thing lol
Soon I'm going to be on model1324 just through etx testing lol

sfjuocekr, 2021-07-09:
I've reported this to OpenTX in the past, but they were not willing to change it.

Maybe when the models are read it could just parse the entire directory and name them according to UUID.bin?

pfeerick, 2021-07-10:
That' s one way to do it. The other is to simply fill in the gap when chosing the next number. Was this an an issue - I'm curious as to the reasoning (of any was given) - as all I can think of is perhaps something to do with synchronization - either with the `models.txt` index file or perhaps some other edge cases.

sfjuocekr, 2021-07-13:
Parsing a directory to look for an empty "slot" is one way to do it, it does introduce extra code to manage models.

I don't think it is an issue, it just does not look nice to have a whole bunch of modelXYZ.bin files that are unused after deleting a model.

Mantamatt1, 2021-07-13:
And the fact that I match the model number with receiver number so I'm not
aimlessly looking for an unused ex number and it messes with my OCD lol

gagarinlg, 2025-01-24:
fixed in current nightlies

closed 2025-01-24 by gagarinlg