EdgeTX/edgetx #367: No rx number duplicated warning

Mantamatt1, 2021-06-28:
Tx16s
Rc3
When binding to a receiver there is no warning that an rx number is already in use

pfeerick, 2021-06-28:
Ouch! Repeatable here with also with TX16S, 2.4.0-RC3. X9D+ and T-Lite show the warning.

Mantamatt1, 2021-06-28:
Ok I'll take a look at my imported models! Maybe somthing went wrong my end

pfeerick, 2021-06-28:
I don't think so... this will most likely be a colour screen radio specific bug.

Mantamatt1, 2021-06-28:
Warning message pops up on otx with tx16s but not on etx so not a colour screen related problem

pfeerick, 2021-06-29:
Sorry to be blunt - but you misunderstand what is mean by "it's a colour screen problem" - OpenTX and EdgeTX support both black & white and colour screen radios, and almost all of the UI code is completely different between interface types. Hence there can be a problem that effects the colour screen user interface, that doesn't effect the black and white radios. This apperas to be one case - EdgetX 2.4.0-RC3 shows the rx number warning on the T-Lite and X9D+ - both B&W screen radios - indicating this is a colour screen radio interface issue. It has nothing to do with a regression in behaviour from OTX, especially as it could be a regression inherited from the 2.4.0 OTX branch. 

It just means that when we look at the cause - it's a matter of looking at the colour screen specific user interface code, rather than looking lower down to see if it's a problem in the underlying receiver id code, since B&W screen radios 'appear' to be unaffected.

closed 2022-09-02 by pfeerick